feelfreelinux/octo4a #187: Error Installing Octolapse Plugin

jjpl2001, 2022-04-16:
Hello. I have problems installing Octolapse on android devices.
I installed Plugin Extras

I tested Android 9 and installed normally on Android 12
Is there something I should install before installing Octolapse?
[plugin_pluginmanager_console_android9.log](https://github.com/feelfreelinux/octo4a/files/8500496/plugin_pluginmanager_console_android9.log)

WRH0530, 2022-04-19:
I have the same problem

WRH0530, 2022-04-19:
In the app, in the settings menu, there is an option called "install plugin extras"

jjpl2001, 2022-04-19:
> 

I had try that many times without success, I made clean installs of the app several times.

What I can see in the install terminal is that there are a few errors during the install process.

jjpl2001, 2022-04-19:
> In the app, in the settings menu, there is an option called "install plugin extras"

I had try that many times without success, I made clean installs of the app several times.
What I can see in the install terminal is that there are a few errors during the install process.

bgross83, 2022-04-23:
Hello, same problem here on Xiaomi Mi 10 Lite with Android 12.
I get the error:

error: command 'gcc' failed: No such file or directory

HunterDarkwolf, 2022-04-29:
The core of the issue for this is that OctoLapse requires the Python 3 dev kit to be installed.  It would help if it came bundled, especially since I would suspect that not a lot of people using this app are going to be able to work out how to install that on their phone.  Personally I'm currently attempting to install it as a downloaded plugin.  Because I'm not one of the ones software capable to install it through something like SSH to my phone.

JustJerem, 2022-05-03:
Yeah, same problem on a Samsung Galaxy S8.
Did you found any work around @HunterDarkwolf  ? :)

alufers, 2022-10-07:
[image]

Yes! Please use the "Install plugin extras" option before installing any plugins.

closed 2022-10-07 by alufers

kpostigo, 2023-05-02:
> Personally I'm currently attempting to install it as a downloaded plugin.

Were you able to get this to work?