feelfreelinux/octo4a #188: Camera streaming eventually kills server (overheat protection?)

CHROM3DOME, 2022-04-22:
I am running Octo4A on two Samsung phones, a Note 9 and a Galaxy 8. On both, if I leave a camera stream or preview open, the phone will eventually stop charging and shut off. I suspect this is from overheating on the phone causing charging to cease. A feature to kill webcam streaming would be ideal, either when charging stops or system temp reaches a certain threshold.

Side note: Octo4A kicks posterior, can't wait to see it on all my spares!

mathiasludvigsson, 2022-04-23:
I have the same problem with my Xperia XZ3.

mathiasludvigsson, 2022-04-23:
I have the same problem on my Xperia XZ3.

Grover49, 2022-05-10:
I have mine on a 100W PD supply and it does the same thing.

Grover49, 2022-05-10:
I was wondering if ther could be a plugin made for Octo4A to monitor the battery and report charge levels and maybe even perform tasks based on parameters, pause prints, or stop the camera server.

Ronin0000, 2022-05-21:
The same thing happens to me with my 2 Pixel 2s when you leave the camera on for a long print the server eventually crashes because of the phone overheating.

ghostKeyboard, 2022-06-03:
My Pixel 4 XL will overheat and shut down after about 10 minutes while viewing the camera stream.

alufers, 2022-10-07:
Unfortunately there is nothing we can do, you can try setting the FPS limit in the camera settings to try to fix this problem.

closed 2022-10-07 by alufers