feelfreelinux/octo4a #190: ​ Octopi4 IP Address 0.0.0.0:5000?!?!?!?! HELP

Ketzukhan, 2022-04-24:
Downloaded Octopi4a on my Android (s8) Phone on WIFI its fine.. I get a proper IP in octoprint. But the moment I try to run it via ethernet cable (HUB adapter) I get the IP Address 0.0.0.0:5000 and obviously cannot connect to my octopi4 through my desktop anymore.... Please help, I've been messing with different settings on the app, settings on phone, and on the internet for 6hrs now.. Even reinstalled the app on the android phone too.

toycz, 2022-05-07:
I have the same exact issue. On WiFi it works perfectly fine but as soon as i connect it via USB -C LAN adapter it gives me server address 0.0.0.0:5000. What is the model of your phone and Android version? I am using Huawei P20 lite with adroid 10.

alufers, 2022-10-06:
As a workaround you can use android settings to see your IP:

https://help.simpletelly.com/article/329-how-to-find-your-android-device-ip-address

alufers, 2023-03-26:
Duplicate of #98

closed 2023-03-26 by alufers