feelfreelinux/octo4a #196: How to fix? ... "Allow OctoPrint for Android to access ARTILLERY_RUBY CCD in FS Mode"

Ltek, 2022-05-05:
I just installed Octa4a an a Samsung A32 (OneUI 4.1)

Every OctoPrint (Android Octo4a) attempts to connect to my Sidewinder X2 there are 2 consecutive phone onscreen popups with the same message...  **"Allow OctoPrint for Android to access ARTILLERY_RUBY CCD in FS Mode?"**

After I answer, Octo connects and works perfectly.

Anyone know how to stop the popups? I'm not seeing Octo4A in Android Settings -> Apps show under Permissions so I cant set any permissions it seems?

The goal is leave the phone / OctoPrint running, not mess with it each time I want to print.

TIA!

stoo85, 2022-05-16:
I have exactly the same problem, I installed team Viewer host on my phone and remotely confirm

Ltek, 2022-05-23:
@feelfreelinux any chance you might fix this?

Ltek, 2022-05-29:
hey guys, looks like @feelfreelinux is ignoring us, not sure why he doesnt even reply.

alufers, 2022-10-07:
I think your printer's VID and PID needs adding to the `device_filter.xml` file. Please post them here and I will add them. 

Please see #225

closed 2022-10-07 by alufers

Ltek, 2022-10-07:
I don't know where to find that. I did a web search and nothing comes up
for Artillery Sidewinder X2 pid or vid

alufers, 2022-10-07:
Hi you can use this app: https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator

Ltek, 2022-10-07:
@alufers thx, the info...

Device Info 
Device Path: /dev/bus/usb/001/032
Device Class: Communication Device (0x2)
Vendor ID:  0483
Vendor Name (reported):  STMicroelectronics
Vendor Name (from DB):  
Product ID:  5740
Product Name (reported):  ARTILLERY_RUBY CDC in FS Mode
Product Name (from DB):  

Additional Info 
Interface #0 
Class: Communication Device (0x2)
Endpoint: #0
Address        : 0x83 (10000011)
Number         : 3
Direction      : Inbound (0x80)
Type           : Interrupt (0x3)
Poll Interval  : 16
Max Packet Size: 8
Attributes     : 00000011
Interface #1 
Class: Communication Device Class (CDC) (0xa)
Endpoint: #0
Address        : 0x01 (00000001)
Number         : 1
Direction      : Outbound (0x0)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 00000010
Endpoint: #1
Address        : 0x82 (10000010)
Number         : 2
Direction      : Inbound (0x80)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 00000010

Ltek, 2022-10-15:
@alufers good?