feelfreelinux/octo4a #198: Klipper 'max_velocity' in section 'printer' must be specified

PixelcrafterEXE, 2022-05-12:
Probably it's user error or hast nothing to do with octo4a but i'm trying to setup my 3d Printer to work with octoklipper on octo4a but it throws a error because of 'max_velocity' not beeing configured even though it is 

My Hardware:
- Phone: Umidigi F2
- Printer: Modified Ender 3
- Mainboard: MKS gen L with Klipper firmware flashed on it

`Trying port /tmp/printer, baudrate 115200`
`Connecting to port /tmp/printer, baudrate 115200`
`Handshake attempt #1 with timeout 2.0s`
`Send: N0 M110 N0*125`
`Recv: // Option 'max_velocity' in section 'printer' must be specified`
`Recv: //`
`Recv: // Once the underlying issue is corrected, use the "RESTART"`
`Recv: // command to reload the config and restart the host software.`
`Changing monitoring state from "Detecting serial connection" to "Operational"`
`Recv: // Printer is halted`
`Recv: !! Option 'max_velocity' in section 'printer' must be specified`
`Send: N0 M110 N0*125`
`Changing monitoring state from "Operational" to "Error"`
`Send: M112`
`Send: N1 M112*32`
`Send: N2 M104 T0 S0*35`
`Send: N3 M140 S0*102`
`Changing monitoring state from "Error" to "Offline after error"`
`Connection closed, closing down monitor`

Here is the full log + config file:
[log and config.zip](https://github.com/feelfreelinux/octo4a/files/8682002/log.and.config.zip)

PixelcrafterEXE, 2022-05-19:
killing the app fixed it

closed 2022-05-19 by PixelcrafterEXE

sheminasalam, 2022-07-03:
> Probably it's user error or hast nothing to do with octo4a but i'm trying to setup my 3d Printer to work with octoklipper on octo4a but it throws a error because of 'max_velocity' not beeing [...]

I am having some issues connecting my ender3 with Octo4a because of not mentioning the correct serial port for mcu. Cna you help me with what all changes you had to make with printer configuration and the octoprint settings to make this work. I mean selecting the serial port and mentioning correct serial in the configuration?