feelfreelinux/octo4a #200: [Feature Request] Set App & Stream ports within Octo4a's Settings

Ltek, 2022-05-13:
It would be helpful if we could change the ports of the Web UI and the Stream URL from Octa4a's setting page on the phone. Thx

DavidAdamsDesign, 2023-02-22:
Yes, please add this - primarily the web UI port. I would really like to be able to run this on 80 and avoid the port at the end of the URL (make things simpler for local DNS).

alufers, 2023-03-21:
> Yes, please add this - primarily the web UI port. I would really like to be able to run this on 80 and avoid the port at the end of the URL (make things simpler for local DNS).

Android forbids binding to anything <1024

DavidAdamsDesign, 2023-03-22:
Interesting article here for non root options:
https://www.geeksforgeeks.org/bind-port-number-less-1024-non-root-access/

alufers, 2023-03-22:
You still need to be root to set the capabilities of the binary or run this daemon from this article. It's impossible to do on an unrooted smartphone.