feelfreelinux/octo4a #204: New Octoprint update breaks Octo4a

wezxwezx, 2022-05-17:
Knew I shouldn’t have updated Octoprint. Server loads on phone on Octo4a, but the browser gives an error message repeatedly.

cp2004, 2022-05-17:
> Knew I shouldn’t have updated Octoprint. Server loads on phone on Octo4a, but the browser gives an error message repeatedly.

This isn't a very helpful bug report. It's no good saying 'Gives an error message', usually maintainers need to know what the error message is.... #205 is an example for you on what is helpful.

feelfreelinux, 2022-05-17:
Looking into the issue!

wezxwezx, 2022-05-17:
> This isn't a very helpful bug report. It's no good saying 'Gives an error message', usually maintainers need to know what the error message is.... #205 is an example for you on what is helpful.

[image]

wezxwezx, 2022-05-17:
> Looking into the issue!

Thank you for working on the problem…I have no idea what I was thinking upgrading without checking around to see if there would be problems! Totally stupid on my part!

drinkspiller, 2022-05-17:
@feelfreelinux — is there a way to force a (re)install of 1.7.x?

simspace-dev, 2022-05-17:
I just picked up a Pixel 3XL and just now installed Octo4a and I am getting the same error ...
[image]

simspace-dev, 2022-05-18:
Figured out how to email myself the logs ... Hope this helps...
[octo4a-logs.txt](https://github.com/feelfreelinux/octo4a/files/8712254/octo4a-logs.txt)

simspace-dev, 2022-05-18:
> @feelfreelinux — is there a way to force a (re)install of 1.7.x?

You can download older .apk's on the releases page:
https://github.com/feelfreelinux/octo4a/releases

wezxwezx, 2022-05-18:
> > @feelfreelinux — is there a way to force a (re)install of 1.7.x?
> 
> You can download older .apk's on the releases page: https://github.com/feelfreelinux/octo4a/releases

If we were to just reinstall an older version it would give us a different IP address than what we had before this correct?

simspace-dev, 2022-05-18:
For those experiencing this problem and find this issue, see [this post](https://github.com/feelfreelinux/octo4a/issues/205) for an update how to resolve it until a fix is released. Specifically [this post](https://github.com/feelfreelinux/octo4a/issues/205#issuecomment-1129345608).

feelfreelinux, 2022-05-18:
Duplicate of https://github.com/feelfreelinux/octo4a/issues/205

Fix released, please install version 1.1.0

closed 2022-05-18 by feelfreelinux