feelfreelinux/octo4a #205: Ocoprint 1.8.x unable to get to web portal

Slart, 2022-05-17:
I was using version 1.7.x successfully, but 1.8 doesn't allow opening of web portal due to permissions issues.  I've included the startup log.

`Got extension Remote web terminal (ttyd)
🐙 2022-05-17 20:15:44,631 - octoprint.startup - INFO - ****************************************************************************** [0m
🐙 2022-05-17 20:15:44,636 - octoprint.startup - INFO - Starting OctoPrint 1.8.0 [0m
🐙 2022-05-17 20:15:44,636 - octoprint.startup - INFO - ****************************************************************************** [0m
🐙 2022-05-17 20:15:44,696 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online [0m
🐙 2022-05-17 20:15:44,697 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working [0m
🐙 2022-05-17 20:15:44,697 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working [0m
🐙 2022-05-17 20:15:44,734 - octoprint.plugin.core - INFO - Loading plugins from /usr/lib/python3.9/site-packages/octoprint/plugins, /root/.octoprint/plugins and installed plugin packages... [0m
🐙 2022-05-17 20:15:49,014 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.3.28) did not pass check, not loading. [0m
🐙 2022-05-17 20:15:49,307 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 34 hook handlers [0m
🐙 2022-05-17 20:15:49,450 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval [0m
🐙 2022-05-17 20:15:49,537 - octoprint.server - INFO - Intermediary server started [0m
🐙 [31m2022-05-17 20:15:49,547 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default [0m
🐙 2022-05-17 20:15:49,949 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /root/.octoprint/uploads... [0m
🐙 2022-05-17 20:15:49,957 - octoprint.filemanager.storage - INFO - ... file metadata for /root/.octoprint/uploads initialized successfully. [0m
🐙 2022-05-17 20:15:49,969 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')") [0m
🐙 2022-05-17 20:15:49,970 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')") [0m
🐙 2022-05-17 20:15:49,971 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')") [0m
🐙 2022-05-17 20:15:49,972 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')") [0m
🐙 2022-05-17 20:15:49,973 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')") [0m
🐙 2022-05-17 20:15:49,973 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')") [0m
🐙 2022-05-17 20:15:49,974 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')") [0m
🐙 2022-05-17 20:15:49,975 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')") [0m
🐙 2022-05-17 20:15:49,979 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')") [0m
🐙 2022-05-17 20:15:49,984 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')") [0m
🐙 2022-05-17 20:15:49,989 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_LIST (needs: "Need(method='role', value='plugin_pluginmanager_manage')") [0m
🐙 2022-05-17 20:15:49,993 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')") [0m
🐙 2022-05-17 20:15:50,002 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')") [0m
🐙 2022-05-17 20:15:50,007 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')") [0m
🐙 2022-05-17 20:15:50,011 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')") [0m
🐙 2022-05-17 20:15:50,015 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')") [0m
🐙 2022-05-17 20:15:52,926 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip [0m
🐙 2022-05-17 20:15:55,588 - octoprint.util.pip - INFO - Version of pip is 20.3.4 [0m
🐙 2022-05-17 20:15:55,603 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no [0m
🐙 2022-05-17 20:15:55,603 - octoprint.util.pip - INFO - ==> pip ok -> yes [0m
🐙 2022-05-17 20:15:55,694 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s) [0m
🐙 2022-05-17 20:15:56,106 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
🐙 |  Action Command Notification Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_notification
🐙 |  Action Command Prompt Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_prompt
🐙 |  Announcement Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/announcements
🐙 |  Anonymous Usage Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/tracking
🐙 |  Application Keys Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/appkeys
🐙 |  Backup & Restore (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/backup
🐙 |  Core Wizard (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/corewizard
🐙 |  Discovery (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/discovery
🐙 |  Error Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/errortracking
🐙 |  Event Manager (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/eventmanager
🐙 |  File Check (2021.2.23) (bundled) = /usr/lib/python3.9/site-packages/octoprint_file_check
🐙 |  Firmware Check (2021.10.11) (bundled) = /usr/lib/python3.9/site-packages/octoprint_firmware_check
🐙 |  GCode Viewer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/gcodeviewer
🐙 |  Logging (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/logging
🐙 |  Plugin Manager (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/pluginmanager
🐙 |  Software Update (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/softwareupdate
🐙 |  Virtual Printer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/virtual_printer
🐙 Prefix legend: ! = disabled, # = blacklisted, * = incompatible [0m
🐙 [31m2022-05-17 20:15:56,119 - octoprint.environment - ERROR - Error while detecting hardware environment [0m
🐙 Traceback (most recent call last):
🐙   File "/usr/lib/python3.9/site-packages/octoprint/environment.py", line 112, in _detect_hardware
🐙     cpu_freq = psutil.cpu_freq()
🐙   File "/usr/lib/python3.9/site-packages/psutil/__init__.py", line 1859, in cpu_freq
🐙     ret = _psplatform.cpu_freq()
🐙   File "/usr/lib/python3.9/site-packages/psutil/_pslinux.py", line 751, in cpu_freq
🐙     max_ = int(cat(pjoin(path, "scaling_max_freq"))) / 1000
🐙   File "/usr/lib/python3.9/site-packages/psutil/_pslinux.py", line 292, in cat
[... cut ...]
🐙   File "/usr/lib/python3.9/site-packages/octoprint/server/api/settings.py", line 34, in _etag
🐙     connection_options = printer.__class__.get_connection_options()
🐙   File "/usr/lib/python3.9/site-packages/octoprint/printer/__init__.py", line 71, in get_connection_options
🐙     "ports": sorted(comm.serialList(), key=natural_key),
🐙   File "/usr/lib/python3.9/site-packages/octoprint/util/comm.py", line 201, in serialList
🐙     with os.scandir("/dev") as it:
🐙 PermissionError: [Errno 13] Permission denied: '/dev'

 
	
`

cjmorri1018, 2022-05-17:
I am seeing the same issue.

drinkspiller, 2022-05-17:
Duplicate of #204

koprda, 2022-05-17:
problem is in new version optoPrint 
src/utils/comm.py on line 201 os.scandir ("/dev")

quick fix:
- comment or delete lines 201-204 in comm.py and reinstall
- I used ssh in okto4a

ssh root@[ip] -p 8022
cd ~/OctoPrint-OctoPrint-f430257
sed -i '201,204d' src/octoprint/util/comm.py
pip install .

... restart server

cp2004, 2022-05-17:
> quick fix:
> - comment or delete lines 201-204 in comm.py and reinstall

Fairly sure that would mean that it didn't list any serial ports, which seems fairly integral to OctoPrint's functions...

drinkspiller, 2022-05-17:
@feelfreelinux  — just curious, does octo4a proactively test against [Octoprint release candidates](https://github.com/OctoPrint/OctoPrint/tags)?

koprda, 2022-05-17:
> > quick fix:
> > 
> > * comment or delete lines 201-204 in comm.py and reinstall
> 
> Fairly sure that would mean that it didn't list any serial ports, which seems fairly integral to OctoPrint's functions...

of course in linux if you have the right you can scan /dev

octo4a (android) does not allow scan /dev and uses a "special" serial port
OctoPrint Settings > Serial Connection > Additional serial ports: "/dev/ttyOcto4a"

mine is working this way ;)

@octoprint dev --- pls use "try except" for os.scandir ;)

cp2004, 2022-05-17:
> of course in linux if you have the right you can scan /dev
> octo4a (android) does not allow scan /dev and uses a "special" serial port
OctoPrint Settings > Serial Connection > Additional serial ports: "/dev/ttyOcto4a"
> mine is working this way ;)
> @OctoPrint dev --- pls use "try except" for os.scandir ;)

Thanks - I forgot Octo4a added a custom additional serial port.

On my install of Octo4a, it works just fine. We'll keep monitoring and look at what can be fixed from OctoPrint's end.

As a side note - there's been **5 release candidates across 2 months** for people to test this release. And then the same day it releases, it turns out that maybe nobody using Octo4a decided to test the update and all of a sudden it's blown up on them. Yes, I'm a bit annoyed that a) I couldn't reproduce the issue in my own tests and b) not enough people are testing the release candidates in that case.

feelfreelinux, 2022-05-17:
> Thanks - I forgot Octo4a added a custom additional serial port.
>
> On my install of Octo4a, it works just fine. We'll keep monitoring and look at what can be fixed from OctoPrint's end.
>
> As a side [...]

Most likely the fault of my neglect of this project, will try to test RCs with newer releases. I have a fix in mind right now, just trying to figure out how to handle it without requiring a full reinstall of the app.

feelfreelinux, 2022-05-17:
> Most likely the fault of my neglect of this project, will try to test RCs with newer releases. I have a fix in mind right now, just trying to figure out how to handle it without requiring a full [...]

on the other side, maybe handling it on octoprint's side would help with that. I'm not sure if i shouldn't temporarily point to my own fork instead.

cp2004, 2022-05-17:
I think some better error handling here is required on OctoPrint's side - previously the code used `glob.glob`, which is a higher level API than the `os.scandir` function, and a peek at the Python source code suggests it was doing the error handling for us while still eventually trying to use `os.scandir`. The permissions error should be handled and not cause an unrecoverable error.

It's not just up to you to test release candidates - it's the responsibility of anyone using Octo4a for their OctoPrint install. If they aren't represented in the testing, then nobody will know if it works or not.

wezxwezx, 2022-05-17:
> I think some better error handling here is required on OctoPrint's side - previously the code used `glob.glob`, which is a higher level API than the `os.scandir` function, and a peek at the Python [...]

I was not aware of the need for us to be testing candidates, but am willing to do this going forward. Though I’m not very well versed in the particulars of how to do this? Would there be info here in the GitHub on how to do that and when it would be needed?

cp2004, 2022-05-17:
> I was not aware of the need for us to be testing release candidates, but am willing to do this going forward if I could be helpful…Though I’m not very well versed in the particulars of how to do this? Would there be info here in the GitHub on how to do that and when it would be needed?

Every time a new release candidate is published, you get a notification (unless you've turned them off) in OctoPrint's UI that contains all the details and instructions of how to update by switching in the software update section.

cjmorri1018, 2022-05-17:
I have this same trouble in my Day Job getting people to really test something, then scream the loudest when it breaks on day 1,

I don't pretend to be an expert, only having a 3D Printer and/or OctoPrint for for about 2 weeks.  I only know enough to be dangerous, but I am glad to contribute going forward to try to "Break it early".   I will look out for those notifications, I am pretty good at breaking things.

I figured it would be easier to deal with any pain in upgrading before I got too many dependencies and things connected.

wezxwezx, 2022-05-18:
> Every time a new release candidate is published, you get a notification (unless you've turned them off) in OctoPrint's UI that contains all the details and instructions of how to update by switching [...]

Ok, no problem…I don’t have announcements disabled and whenever I’ve seen an upgrade available I’ve installed it…I’ve just never had any issues before?

simspace-dev, 2022-05-18:
> 
@koprda, thanks your steps to fix the issue worked for me!

simspace-dev, 2022-05-18:
@wezxwezx each command is separate and needs to entered one at a time. 

1) ssh root@[ip] -p 8022
2) cd ~/OctoPrint-OctoPrint-f430257
3) sed -i '201,204d' src/octoprint/util/comm.py 
4) pip install .

It may be easier to use a terminal on your computer. Are you a Windows or Mac user?

wezxwezx, 2022-05-18:
> @koprda, thanks for your steps to fix this issue. It worked great for me!

I downloaded an SSH client app and unbelievably (because I have no idea what I’m doing) I have been able to log into the server, but I ran the code listed and I just get errors…can you help me by any chance? This is what I typed: cd ~/OctoPrint-OctoPrint-f430257
sed -i '201,204d' src/octoprint/util/comm.py
pip install .

wezxwezx, 2022-05-18:
> 

Windows…man I suck at this!

mom logged into the server in this ssh app and I have 
localhost:~#

I typed in the code line starting with “ssh root…etc” but it says ssh not found. Then I tried, since I’m already logged into the server and I thought maybe that first line is to login which I already did, doing the next line “cd…etc” but it says “no such file or directory”

wezxwezx, 2022-05-18:
> @wezxwezx each command is separate and needs to entered one at a time.
> 
> 1. ssh root@[ip] -p 8022
> 2. cd ~/OctoPrint-OctoPrint-f430257
> 3. sed -i '201,204d' src/octoprint/util/comm.py
> 4. pip install .
> 
> It may be easier to use a terminal on your computer. Are you a Windows or Mac user?

I downloaded the PuTTY SSH program...This is what keeps happening..
[image]
.

drinkspiller, 2022-05-18:
>    1. ssh ***@***.*** -p 8022

Remove the brackets around the IP address:
ssh ***@***.*** -p 8022

wezxwezx, 2022-05-18:
> Remove the brackets around the IP address: ssh ***@***.*** -p 8022
> […](#)

I tried that...keep saying "ssh not found"

wezxwezx, 2022-05-18:
BTW Thanks for trying to help me you guys...

Guardian925, 2022-05-18:
Think they'll put out another update soon to fix the issue? I don't have much faith in me trying to figure out where to add certain codes.

wezxwezx, 2022-05-18:
> Think they'll put out another update soon to fix the issue? I don't have much faith in me trying to figure out where to add certain codes.

I hope so, I'm not COMPLETELY clueless on this terminal stuff...but I really don't have much of an idea of what I'm doing LOL

wezxwezx, 2022-05-18:
Just keeps doing this...???

[image]

parrotsgain, 2022-05-18:
because your already loged in now do a cd /root and ls -lah , you will find a folder called Octoprint-****** thats the folder you should sed in

wezxwezx, 2022-05-18:
Well, I give up for the moment...I can login to the Windows PuTTY SSH program with my ip "192.168.1.119" and port "8022" and then it asks for username and I can login with either "root" or "Octoprint" and then my password I set up in Octo4a. That brings up localhost:~# 

but no matter what I get the same response...

[ssh root@192.168.1.119 -p 8022]   RESPONSE    [-ash ssh: not found]

I even tried the second command line and also nothing...

[cd ~/OctoPrint-OctoPrint-f430257ssh]    RESPONSE    [-sh: cd: can't cd to /home/octoprint/Octoprint-Octoprint/f430257: No such file or directory]

parrotsgain, 2022-05-18:
well ofcourse not cause the f4***** part is unique to each installation  do `ls -lah  /root `

wezxwezx, 2022-05-18:
> 

Ok, I'm in WAY over my head here...thank you for posting this and I'm trying very hard to follow what you're saying.  I found the octoprint-****  file

wezxwezx, 2022-05-18:
I entered this:  apk update && apk add nano && nano /root/OctoPrint-OctoPrint-5a1b07
e/src/octoprint/util/comm.py

And got this: fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/aarch64/APKINDEX.tar.gz

Have no idea what to do now...?

wezxwezx, 2022-05-18:
[image]

wezxwezx, 2022-05-18:
Am I supposed to open the APKINDEX file and then substitute the code you wrote?

parrotsgain, 2022-05-18:
did you do a apk add nano 
nano /root/OctoPrint-OctoPrint-5a1b07/src/octoprint/util/comm.py then do those

wezxwezx, 2022-05-18:
> did you do a `apk add nano ` `nano /root/OctoPrint-OctoPrint-5a1b07/src/octoprint/util/comm.py` then do those

OK, I thought that was all one command. Sorry.  I did "apk add nano" and now it says (1/1) Installing nano (5.7-r2)
0%

But its just staying 0% and not doing anything...?

simspace-dev, 2022-05-18:
> besides the sed -i '201,204d' wont do anything,

The `sed` line was the key to resolving this issue for me. I had the same exact problem @wezxwezx is having. I executed those four CLI commands and now Octo4a is working as expected. I can print and control my 3D printer remotely with no issues. 

@parrotsgain, I'm not suggesting your fix won't work. I'm just clarifying that deleting lines 201-204 does work as well and that suggested fix may be easier for @wezxwezx to implement. 

@wezxwezx, if you can successfully SSH into your device, I suggest trying the less complex fix first that was [explained in the post above](https://github.com/feelfreelinux/octo4a/issues/205#issuecomment-1129345608).  

- cd ~/OctoPrint-OctoPrint-f430257
- sed -i '201,204d' src/octoprint/util/comm.py
- pip install .
- restart the server (or just reboot the Android device and then open Octo4a).

Bedtime! I hope you get it working. Happy printing.

wezxwezx, 2022-05-18:
Ok, that did something!
[image]

wezxwezx, 2022-05-18:
Thank you to both of you for trying to help me...

parrotsgain, 2022-05-18:
> suggest trying the less complex fix first that was [explained in the post above](https://github.com/feelfreelinux/octo4a/issues/205#issuecomment-1129345608).
>
> * cd ~/OctoPrint-OctoPrint-f430257
> * [...]

i tried your method first but my installer was corrupted so half files were 1.8 half 1.7 and this  rendered it unusable hence why my solution is to back port some of the 1.7 code.

wezxwezx, 2022-05-18:
Hmmm...ok, restaryed Octo4a and now it just says "Octoprint is starting" endlessly and never connects...uh oh...

wezxwezx, 2022-05-18:
> did you do a `apk add nano ` `nano /root/OctoPrint-OctoPrint-5a1b07/src/octoprint/util/comm.py` then do those

I'm sorry...I'm just not understanding.  I've run this command and it just sits at 0% and never moves. If I go ahead and add the "nano /root/OctoPrint-OctoPrint-5a1b07e/src/octoprint/util/comm.py" command, nothing happens...?

[image]

wezxwezx, 2022-05-18:
What would happen if I just hard reset the phone and installed Octo4a from scratch?  Would it still be Octoprint 1.8 or would it be 1.7 and work?

wezxwezx, 2022-05-18:
I have figured out some things: After running the 4 command lines and discovering that Octo4a would not even connect anymore, I did a hard reset on the phone and reinstalled Octo4a. It now will connect. Of course that’s with a new ip, but it’s really not that big of a deal. So I opened the ssh terminal and ran the directory command and lo and behold, the Octoprint-Octoprint-**** file is now the same f430257 that simspace-dev had referenced. Apparently on a new install that file is always named f430257? My old install was a completely different file - 5a1b07e. So I ran the 3 command lines:
• cd ~/OctoPrint-OctoPrint-f430257
• sed -i '201,204d' src/octoprint/util/comm.py
• pip install .
And then restarted the phone. The Octoprint browser came right up…no error message. I’m restoring my settings from backup now.

For whatever reason, my old install would not work with those changes, but a fresh install did…?

simspace-dev, 2022-05-18:
> i tried your method first but my installer was corrupted so half files were 1.8 half 1.7

@parrotsgain, that's stinks! Glad you figured out another way to get around the issue.

@wezxwezx, glad you got it working!

feelfreelinux, 2022-05-18:
https://github.com/feelfreelinux/octo4a/releases/tag/1.1.0
Released, apk should be available there shortly.

Thanks a lot for the feedback!

wezxwezx, 2022-05-18:
> https://github.com/feelfreelinux/octo4a/releases/tag/1.1.0 Released, apk should be available there shortly.
> 
> Thanks a lot for the feedback!

Thanks for this!…should I just install this over top of my current installation? I did get it working with the work-around given here in this thread.

wezxwezx, 2022-05-18:
> > i tried your method first but my installer was corrupted so half files were 1.8 half 1.7
> 
> @parrotsgain, that's stinks! Glad you figured out another way to get around the issue.
> 
> @wezxwezx, glad you got it working! I should have commented that the `pip install .` takes a while. I updated my comments for others who may stumble across this issue too.

Hey, THANKS TO YOU BOTH for helping me last night!  I'm sure that was frustrating for you because I'm so out of my element with trying to run codes and such...appreciate your patience!

cjmorri1018, 2022-05-18:
@feelfreelinux 

Thanks very much for the quick turnaround.  So far things look good.

wezxwezx, 2022-05-19:
> @feelfreelinux
> 
> I installed the new 1.1.0 version several hours ago, and rebooted several times while testing, and playing all seemed fine
> 
> The issue re-emerged briefly with same/similar message citing \dev when attempting to reboot after the re-install of a plugin. I had successfully uninstalled PSU Control plug in, rebooted successful, reinstalled PSU Control plugin and attempted restart displayed same error on Windows pc browser, the android 10 device browser interface displayed a message that I didnt catch. A restart of the app, and both browsers (pc and phone) seemed to clear it up.
> 
> In the process of attempting to recreate it I did a few uninstall and reinstalls of the same plug in, seemed fine, but when I did an uninstall & clean up data, then a re-install and reboot the issue returned.
> 
> [image] [image]

I have periodically and seemingly randomly gotten that “Loading Failed” error message on my iPhone ever since I began using Octoprint. Reloading the page has always made it go away…except of course yesterday with the 1.8 issue LOL I’ve gotten it a couple of times with 1.1.1 but it has always resolved with a browser refresh. I get a lot of weird errors but only on the iPhone (6S+)…maybe because the phone is so old? I get a lot of mostly Octolapse loading errors and also settings loading errors, but ONLY on the phone…never on my desktop (Windows 7). I just dismiss them and they never seem to cause any trouble. If it’s an Octolapse error and I want to use Octolapse I just refresh the browser and it resolves.