feelfreelinux/octo4a #208: versionCode increase

IzzySoft, 2022-05-18:
Hi, looks like you forgot increasing `versionCode` with the last release. v1.1.0 still has `1000004` (from v1.0.4). So Android won't consider it an update but the same version :wink: Just thought I'd leave a note.

Hm, looks like it also states it's v1.0.4, but the APKs differ… [Oops](https://github.com/feelfreelinux/octo4a/blob/master/app/app/build.gradle.kts#L16), indeed…

PS: Downloads are also [available here](https://apt.izzysoft.de/fdroid/index/apk/com.octo4a) in my repo, so users can keep the app updated easily – provided `versionCode` is increased with each release :wink:

feelfreelinux, 2022-05-18:
Fixed in new release :) Thanks!

IzzySoft, 2022-05-18:
Great, thanks! Btw: if you wish to link to your app's presence in my repo, feel free to [pick the corresponding badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) :wink:

feelfreelinux, 2022-05-18:
> Great, thanks! Btw: if you wish to link to your app's presence in my repo, feel free to [pick the corresponding badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) 😉

Added, thanks a lot!

IzzySoft, 2022-05-18:
Great, thanks! But could you also change the link target to https://apt.izzysoft.de/fdroid/index/apk/com.octo4a please?

IzzySoft, 2023-10-28:
@feelfreelinux looks like it's not fixed: we had `versionCode: 1001002` now for 3 times (last one from June 2023, so more than 1 year after it was supposedly fixed) – and your `build.gradle.kts` [still has that code](https://github.com/feelfreelinux/octo4a/blob/master/app/app/build.gradle.kts). Again, nobody received any update notifications :cry: Do we get that tackled one day?