feelfreelinux/octo4a #211: 1.0.4-1.1.1 crashing on install, 1.0.3 install quickly jumps to 150%, relaunching app reveals installation corrupted

goosoodude, 2022-05-20:
Hi! I'm running a moto g pure (model XT2163DL - locked to TracFone with a locked bootloader) specifically so I can use octo4a to monitor and remotely start prints. I've tried all versions starting from 1.1.1 all the way to 1.0.4, and the installation will crash every time around the 50-75% mark, and obviously, I cannot get the log output to see where it may be failing. 1.0.3 however does not crash, but it doesn't reveal much. Alpine is able to download, and from what I can tell the output seems to be normal. The progress does almost instantly progress to 150% with the continue button being unable to be interacted with. The log can be seen below:

💻 No bootstrap detected, proceeding with installation
💻 Downloading bootstrap 0.1.8
💻 Bootstrap extracted, setting it up...
🐸 add-user.sh
🐸 fake_proc_stat
🐸 install-bootstrap.sh
🐸 ioctlHook.so
🐸 root
🐸 root-pre5
🐸 rootfs.tar.xz
🐸 run-bootstrap.sh
🐸 bootstrap ready, run with run-bootstrap.sh
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Welcome to Alpine!
🐸 
🐸 The Alpine Wiki contains a large amount of how-to guides and general
🐸 information about administrating Alpine systems.
🐸 See <http://wiki.alpinelinux.org/>.
🐸 
🐸 You can setup the system with the command: setup-alpine
🐸 
🐸 You may change this message by editing /etc/motd.
🐸 
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: apk: not found
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: can't create /etc/ssh/sshd_config: nonexistent directory
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: ssh-keygen: not found
💻 Bootstrap installation done
💻 Bootstrap installed
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: bash: not found
🐸 /bin/sh: curl: not found
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: curl: not found
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 unzip: can't open octoprint.zip[.zip]
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 /bin/sh: cd: line 1: can't cd to Octo*: No such file or directory
💻 Dependencies installed

There's no firewall that could be blocking any downloads, no restrictions. Just to be 100% safe I tried installing with a VPN active, but that didn't help. The only thing I can think of is maybe it's something to do with the weird restrictions the carrier might've placed on the phone, but I really don't think that would cause the installation to fail. And ideas would be appreciated.

goosoodude, 2022-05-20:
Small update: I've installed 0.0.5, which DID actually fully install and allowed OctoPrint to start. Though, upon opening the web interface, I was met with a "loading failed" message, though not going to worry about trying to fix that as I was only trying to test lower versions. I'm guessing this has something to do with Alpine as I've also attempted to run 1.0.0, and the outcome is exactly the same as 1.0.3.

Are there any system settings I may need to enable/disable? This seems to be an issue with Alpine downloading but not actually properly running or something. Any suggestions are welcome at this point.

miroan81, 2022-06-05:
Flash your phone with different android version, Ive had the exact problem on Xperia and that fixed the issue. Check for FW on XDA forum

Deadstic, 2022-06-17:
Same phone and same problem tried everything you have nothing is working.

LD40, 2022-07-11:
Same problem here : have tried 1.1.1 on several android devices and versions (from 5. to 7.0) ; same problems than @goosoodude : lot of warnings, for bindings, inflating, ... that seem relative to access rights.

As for him, not a network or FW issue, but clearly some problem in the installation process (some early element that have changed ?)

dcjweller, 2022-07-11:
Same issue with me on Android 4.4!
I'm using an old LG G Pad, so I'm stuck at 4.4 
Would be great for Octoprint to work on this version without needing to flash a new Android version

eredwan, 2022-07-19:
I am having similar error log but with more errors. On an LG K7 running android version 5.1.1.
I am not on a firewall (except the default router firewall) or VPN or anything restrictive on my home network. 
I have even tried hot spotting on my other phone using the cell data network. Still the same issue.

tycho94, 2022-07-29:
After experiencing the same issue, I got my phone to work by switching from a Galaxy S6 stock Android 7 to lineage OS 17.1 - Android 10. It requires rooting, a custom bootloader and custom OS but since most of us are doing this on an old device you probably won't void your already expired warranty.

goosoodude, 2022-07-29:
Alright, after doing a lot of testing, it's almost certainly something to do with the highly locked-down state of Tracfone's devices. If you can, unlock your bootloader and flash a stock ROM. But most Tracfone devices have locked bootloaders that cannot be unlocked.

Since some people may be buying phones to use as print managers since Raspberry Pis are extremely expensive, I recommend returning any TFW-based phone you might've purchased and instead buying a phone meant to be used on Boost Mobile's network. Walmart sells them, and their cheapest phone is a $40 TCL 20XE. It works fine, its camera is decent, and it can be used with a USB-C hub to both charge and interface with the printer.

If you're using an older phone you have lying around... sorry to say, it seems like most Tracfone Wireless-locked phones are incompatible with octo4a due to the restrictions in the firmware. See the previous paragraph if you really need a device to use for Octoprint.

closed 2022-07-29 by goosoodude

Fabe3, 2022-09-22:
I'm having the same issue with a Samsung Galaxy s7.  I've tried v1.1.1, v 1.1.0, and v1.0.4 on a factory reset galaxy s7 over both wifi and wifi disabled cellular data downloads.  I'm not sure how to change OS on phone to try that, so I'm at a bit of an impasse on how to proceed.  Might have to break down and buy the Raspberry Pi.

CristianG5401, 2022-12-12:
This fix works for me https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359