feelfreelinux/octo4a #212: issue with SSH

Unclyboi, 2022-05-25:
So im trying to setup klipper for my voron 0.1, and none of the commands on the voron website will work

[image]

any help would be greatly appreciated

parrotsgain, 2022-05-26:
different distros use different package mangers and usually different package names.
apt is for debian/ubuntu based
apk is for alpine. apk add **** or apk del *** and no sudo  
https://wiki.alpinelinux.org/wiki/Package_management

your home is in /root and /root/Octoprint-*****

sheminasalam, 2022-07-03:
> 

Did you find get klipper to work? Do you have any guides?

alufers, 2022-10-06:
Marking as solved by @parrotsgain

closed 2022-10-06 by alufers