feelfreelinux/octo4a #214: OnePlus 3 charging leak with Artillery Genius

Gby56, 2022-05-25:
Just thought I would give some feedback and history for anyone searching about oneplus support!
It works perfectly when it can, but the battery charging makes this pretty useless for now, not being a proper 24/7 server.

I bought the Y cables and hoped to be able to charge and OTG at the same time, I'm simply using an apple usb c to usb A female, then I put the Y cable, which splits between the printer and an apple charger.
There is visibly no protection on the printer's USB port for "backwards" charging because it boots up and beeps when I plug the phone in, even without the apple charger !

The big problem is that the Y cable is useless, because the oneplus 3 doesn't seem to want to charge when I turn on the charger, probably because it's leaking the charger to the printer, and the printer can never shut down until the phone is shut down, which makes quite a complex setup hahah

kilimar, 2022-05-30:
This isn't necessarily OnePlus issue.  As you said, it's the OTG cable.  You probably should post the brand, model, etc of the cable so that folks can avoid it.  I got the same issue with my Y cable on my Moto E4 Plus XT177.  I purchased my OTG Y-cable over 4 years ago, so being the cheap cable from eBay, made and delivered from China, the current ones' internals are probably different; never the less, the photo is here for reference.
[image]

One possible issue is that the newer phones are using Power Delivery or Quick Charge and the Y-cables are only delivering up to 7.5 watt instead of 9W+

Gby56, 2022-05-30:
@kilimar yeah it's more of a cable issue, I wonder how we could add a diode to prevent the phone from sending power to the printer... but it might just break the whole communication

Gerigot, 2022-05-31:
I use a Y cable from china and it's working perfectly ... what I've done is cut the 5v wire of the usb cable used between the phone and the printer then I make sure the charger is not too new because new models of charger (with fast charging, power delivery or quick charge) communicate through data's usb wire and may cause interference (at least to me it was always disconnecting).

kilimar, 2022-05-31:
> I use a Y cable from china and it's working perfectly ... what I've done is cut the 5v wire of the usb cable used between the phone and the printer so that the printer will not turn on with the power of the phone. 

That's a good work-around that forgot to mention.   Instead of cutting wires, you can also put tape on one the 5V pin of the USB-A.  Any tape will work, thinner is better; polyimide tape (aka Kapton) is probably the best.  Cut a thin strip like 1-2mm by 10mm.  Slide it over the 5V pin, fold it down, and carefully plug it into the adapter cable, then plug the other end to the printer.  Then verify that the power doesn't come-on on the printer.

> Then I make sure the charger is not too new because new models of charger (with fast charging, power delivery or quick charge) communicate through data's usb wire and may cause interference (at least to me it was always disconnecting).

This is a good point, I'll have to see if I can find a dumb charger that doesn't use PD or QC protocols.

Gby56, 2022-05-31:
Thanks a lot for the great ideas ! I was going crazy with my soldering diode ideas lol, this should be added in the wiki documentation!

Gerigot, 2022-05-31:
> Thanks a lot for the great ideas ! I was going crazy with my soldering diode ideas lol, this should be added in the wiki documentation!

It is written in the Troubleshooting Guide: 
[https://github.com/feelfreelinux/octo4a/wiki/Connecting-to-printer-via-OTG,-simultaneous-charging](url)

Gerigot, 2022-05-31:
> > I use a Y cable from china and it's working perfectly
> 
> Can you post a photo or link to the cable/adapter that you are using? Is it USB C or Micro?

I bought it on Aliexpress if you search something like "pd charging port otg cable"
here is a pic
[image]

as i said above do not use a PD or better a fast charger of any kind or at least avoid using something that needs some sort of communication between the phone and the charger, because I think, with those cheap OTG cables there will be interference and I had myself some problems with disconnection using a PD charger. 

Now I'm using an old USB type A charger with a Type A -> type C cable and for precaution I added some electric tape on the data pins to avoid any interference.

maybe I'm wrong but it's working fine since i've done those modification.

A recap of my set-up:
connected a y cable (with otg ) like the one in the picture above to the phone running octo4a.
from that cable i connected the printer cutting out the 5v cable ( it's also enough to use electrical tape to isolate the pin in the connector but I had a cable that I could easly cut out).
On the oter side I connected a cable usb-c - usb-A to the charger, covering the two central pins in the type A usb connector (data pins) to avoid any interference.

I hope I was helpful

Gby56, 2022-06-05:
> > Thanks a lot for the great ideas ! I was going crazy with my soldering diode ideas lol, this should be added in the wiki documentation!
> 
> It is written in the Troubleshooting Guide: [https://github.com/feelfreelinux/octo4a/wiki/Connecting-to-printer-via-OTG,-simultaneous-charging](url)

@Gerigot I don't see that technique of cutting the power cable or taping it anywhere in the wiki, this would be a good information to give

Gby56, 2022-06-05:
I just tried again with both my Oneplus3 and Xiaomi Mi4 to cut the red power cable from my Y splitters... It seemed to improve a bit, definitely the printer is not turning on anymore with the phone plugged in.... But now the charging seems to act randomly on both phones ! if the printer is disconnected and off, and I turn on the charger, the phone won't charge, but if the phone is off, it seems to be working (I use an apple ipad charger, the charger is good)
I think the phones switch to a weird charging mode once the printer is turned on, and won't allow to be charged as usual afterwards, which means power cycling phones is necessary to get them charging, still :(

Gby56, 2022-06-05:
Update: apparently some folks already worked on that issue for the other OnePlus phones, kernel patching is required... https://forums.oneplus.com/threads/charging-otg.305674/

Gerigot, 2022-06-06:
> @Gerigot I don't see that technique of cutting the power cable or taping it anywhere in the wiki, this would be a good information to give

oh sorry you are right, I've read about this here: https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

alufers, 2022-10-07:
Closing, as this is not a problem with octo4a itself.Feel free to continue the discussion here, though.

closed 2022-10-07 by alufers

jkmathew, 2023-05-29:
Hey @Gby56,
I am also in the same situation, I tried to use my old Oneplus 6T with Octo4A. Everything works except I cannot charge the phone while connected to the printer. I saw the forum posts that suggests to patch Kernel. I am completely lost in this step. Did you manage to do that? I mean patch Kernel to work charge and connect to the printer at the same time.

Gby56, 2023-05-30:
Hi @jkmathew ! I kinda gave up I think, didn't want to bother flashing the kernel and going custom, I have a raspberry pi running octopi instead 😅 It's a shame that this is not simpler, the overall idea of using a phone for a 3d printer remote control + camera is great

jkmathew, 2023-06-21:
Thanks for the update :) 
Yeah, Raspberry Pi is much simpler I think. The main reason I tried to use the phone is Pis are too costly atm and I have an old phone, so thought not to spend more money on it 😄