feelfreelinux/octo4a #217: Expose battery level

Gerigot, 2022-05-28:
Is there a way to expose  through API or some script the battery level status so that, maybe with a plugin, being able to show it to the octoprint page, or connect it to the telegram plugin to have an alert when battery is about to run out.
I had a problem not so long ago that I forgot to close the webcam view in cura and my phone "died" in the middle of a print.

A part of that now It's almost 6 month since I started using Octo4a with my ender 3 v2 and it's working perfectly so I want to say thank you all for the great work.

wjflier, 2022-05-29:
solder charge cable to phone

Gerigot, 2022-05-29:
the problem it is not the charging system in this case, I'd like to have the battery level somehow exposed if it's possible.
My exemple was only a use case, I know there are some other solutions but it would be nice to see the battery level in octoprint.

kilimar, 2022-05-29:
> solder charge cable to phone

Not everyone had the ability to solder (or the equipment or even the tools to open up the phone).  :-)

miroan81, 2022-06-05:
You cant use this and rely only on a phone battery, its useless and impractical. You must either make or buy Y cable that will allow you to charge phone at the same time or soldier buck converter to a battery terminals. Y cable doesn't work with all phones but if it does its the most easiest solution

Gerigot, 2022-06-05:
I know I wrote that my phone is always in charge but the webcam sometimes uses more energy that the charger can give ... And, as i wrote, it's also a supplementary feature that can be used just for showing the charge, mine was only a use case.

mramirkhan, 2022-06-05:
The Y cable isn't supported on many phones. Thankfully it is on mine but I had to install many many kernels until I found a setup that allowed it. It's a real problem, not one I face but never the less i do understand what OP is feeling.

tobiasgraf, 2022-06-21:
I've hacked together a little plugin that exposes the battery level of the phone and can send telegram messages when battery is running low. Have a look here

https://github.com/tobiasgraf/OctoPrint-Octo4a_battery

Gerigot, 2022-09-27:
your plugin is amazing thank you! that's exactly what I needed.
I will close this Issue.

closed 2022-09-27 by Gerigot

davidalim, 2022-10-09:
Anyone know the battery file path for Samsung Galaxy S5?

davidalim, 2022-10-12:
I used ADB to browse to the file and it's actually there and has the battery percent. However still shows INVALID PATH.

davidalim, 2022-10-13:
Yes totally. Didn't know much about these files so assumed it was just my phone or something :)

I'm running this on a Samsung Galaxy S5 with LineageOS 18.1 (Android 11). Not rooted. 

I did just install on my Galaxy S4 (running Pixel Experience 9) and it works! So personally no longer an urgent issue.