feelfreelinux/octo4a #219: I can't connect to the server using browser.

Kaptus2137, 2022-06-04:
This is the first time I am asking for help, but I have a problem.  When I click on the ip on the phone everything works. But i can't connect to the server using  browser. Thank you in advance and best regards.

miroan81, 2022-06-05:
How the IP in an app looks like 192.x.x.x OR 0.x.x.x ? If it starts with 0 then you're not connected to local wifi, if not make sure you put :5000 at the end

Kaptus2137, 2022-06-05:
> How the IP in an app looks like 192.x.x.x OR 0.x.x.x ? If it starts with 0 then you're not connected to local wifi, if not make sure you put :5000 at the end

Its look normal start with 192.168.x.xxx:5000

Kaptus2137, 2022-06-05:
Whent i click on ip in the app the octoprint control panel  is loding and works but when i want to conect to the server typing ip in my browser the site is loding loding and loding and then i get normal browser eror:

[image]

Kaptus2137, 2022-06-05:
When i click on ip in the app the octoprint control panel  is loding and works but when i want to conect to the server typing ip in my browser the site is loding loding and loding and then i get normal browser error:

[image]

Gerigot, 2022-06-06:
Are you connected on the same network of the phone? if yes, as mentioned above, did you put the :5000 at the end of the url? for example, supposing your phone/server ip is 192.168.1.100 you should put: http://192.168.1.100:5000

Kaptus2137, 2022-06-06:
Yup i put :5000. My computer and second wifi router are connect to the main router. Phone is conect to the secont router.

Gerigot, 2022-06-06:
probably the problem lies there the fact that you are using two router try to check the subnetwork or try to see how can you make those two network speak with each other.
The main router and the second router are in the same network but the phone is in the second router's network not accessible from your compoter on the first router's network

Kaptus2137, 2022-06-06:
Thx 
[image]

closed 2022-06-06 by Kaptus2137