feelfreelinux/octo4a #221: Camera Framerate - Battery

GustavoZani, 2022-06-05:
Hi, I have a very old cellphone, that the battery is not so good, could you please develop a lower frame rate for the camera? For exemple, one photo in one minute. I believe that would reduce the quantity of data, and then, would reduce the amount of battery that it consumes.

What do you think?

Thank you in advance

wjflier, 2022-06-06:
hard wire the phone.

GustavoZani, 2022-06-06:
Hello, you mean connect without battery? I have an Samsung s6

wjflier, 2022-06-07:
Yes, I use a Samsung j3.  I'd find a cheaper phone with removable batter.  solder 2 wires +, - to the contact pins on the phone, no battery needed.

GustavoZani, 2022-06-07:
Good. just a question, battery of Samsung S6 seems horrible to have solder processing. 
Is possible to guide me? I don't know how to do it with that battery.

GustavoZani, 2022-06-08:
Look the model here: https://portuguese.alibaba.com/product-detail/Original-Replacement-Battery-EB-BG920ABE-For-1600181357830.html