feelfreelinux/octo4a #224: Step further: Octodash works with octo4a!

xiaoxx970, 2022-06-08:
Since we have our Android phone runs Octoprint, why not do a little more, install Octodash on the phone, so that we can make full use of the touch screen.
I wrote an article about this: https://xiaoxx.cc/use-octodash-on-your-android-phone-en
[image]
Proceed as follows:
- Install [Termux](https://termux.com/) on your phone
- Install the [PRoot](https://wiki.termux.com/wiki/PRoot) distribution for Debian in Termux
- Install Octodash in Debian
- Install Xserver XSDL on your phone
- Test running Octodash under Debian in Termux
- Add Octodash as a service in Termux and enable automatic startup
- Done, use Octodash in Xserver XSDL

Ltek, 2022-07-11:
Thx for posting the how-to, I'm sure some people will do it. Curious why do that (decent amount of work) vs using one of several Octo apps (which work great) posted on the Play store?

xiaoxx970, 2022-07-12:
Honestly I didn't know there's several octoprint's client on the Play store, I do now... and some of them looks nicer than octodash. Well guess I have made things complicated.

mattkerrison, 2022-08-16:
@xiaoxx970 I like the idea of using Octodash, I set it up as per your guide and it works well, the only issue being that the screen never turns off, not by the default android timeout or when Octodash is sleeping. It's the only issue I see as I'm using a phone with an oled screen and don't want burn in. Did you manage to find a way to allow the screen to timeout and sleep?

xiaoxx970, 2022-08-18:
@mattkerrison My phone doesn't automatically lock either when running xserver. I guess it's the software which prevents the system from locking, and I didn't find the relevant option in the software settings. But I actually found that if I don't operate it for a period of time, the content displayed by the xserver will become a black screen. If your screen is OLED, displaying a black screen should also protect the screen. I'm not sure if the black screen is from Octodash or Xserver XSDL though, if your phone doesn't automatically black out (not lock screen) then you could probably check out the Octodash config. Or use [ Android X11 X-server](https://play.google.com/store/apps/details?id=au.com.darkside.xdemo) which I just found it won't prevent system lock, but require change x port to 6000 in debain.