feelfreelinux/octo4a #226: Nuu Mobile R1 does not work.

WriteCodeEveryday, 2022-06-11:
The Nuu Mobile R1 is capable of doing OTG alongside charging but plugging it into the printer causes nothing to happen.

It looks like it might be a hardware issue but yeah, kinda of a bummer.

I thought [it was an octoprint issue and posted logs and everything over there.](https://community.octoprint.org/t/cr-30-on-octoprint4android/44946)

WriteCodeEveryday, 2022-06-11:
[image]

alufers, 2022-10-07:
Hi, thanks for trying octo4a!

Please try connecting an USB mouse first. If a cursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

PS: feel free to ping me if you need this issue to be re-opened

closed 2022-10-07 by alufers