feelfreelinux/octo4a #230: Time lapse aspect ratio

47dahc, 2022-06-21:
First of, Thank you for and awesome app. The only problems I've ran into so far is I can't seem to get time lapse video to record in 16:9 aspect ratio. All octoprint and cameras settings are set to 16:9 but rendered video is 1:1. Control tab has preview in 16:9 though. Running version 1.1.1 on a Google Pixel 2 XL with Android 11.

47dahc, 2022-07-12:
Seems that in order to change the resolution of the timelapses there is a "octopi.txt" file where the setting for resolution is when running Octolapse on a Raspi. I am unbale to find this file in octo4a and can't seem to find a way to change the resolution to anything other than what looks to be 1:1. Unless I'm missing something somewhere, doesn't seem to be possible to change the resolution to 16:9 (1920x1080).

WriteCodeEveryday, 2022-07-20:
Same thing here, quality is terrible here with a 1:1 resolution

FeuerDrachenEinhorn, 2022-08-15:
same for me with the 1:1 ratio

kristoferahlqvist, 2022-10-25:
Same for me, doesn't matter what resolution or camera I choose. Still 1:1 ratio

asharma2040, 2023-01-18:
I am having similar problem. Has anybody able to fix this?

SebastienRamsay, 2024-01-26:
still no fix?

Scanlia, 2025-06-10:
This still seems to be an issue for me