feelfreelinux/octo4a #234: Upload from SuperSlicer HTTP:500 Error and Upload to SD Card Fail

JorgeSDF, 2022-07-03:
Hi, when I am trying to upload gcode from Superslicer, the file transfers but it shows an error HTTP:500. Also, when I want to upload to the sd card, Octoprint shows me a message that says Error Uploading to SD.
I have an Ender 3 V2 with Marlin 2.0 Jyers Firmware.
Samsung Galaxy S7
Original Samsung MicroUSB and OTG Connector

SuperSlicer has encounter an error

Error uploading to print host:
HTTP 500: {"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}

Upload failed
Could not upload the file. Please check octoprint.log for possible reasons.

Octoprint.log

OSError: [Errno 18] Cross-device link: '/tmp/octoprint-file-upload-gmte2f4_.tmp' -> '/root/.octoprint/uploads/Shape-Box.gcode'

alufers, 2022-10-06:
Hi! Please provide the log file. I will l then reopen the issue.

Thanks!

closed 2022-10-06 by alufers

alufers, 2022-10-07:
Duplicate of: #161