feelfreelinux/octo4a #236: Octo4a crash on octoprint installation

0rrsss, 2022-07-07:
[image]
[image]

Permissions are given, but app crashes when attempting to install. I've tryed a few different versions with no success. Phone is a blue view 2 just out of the box with no other downloads.

LojikNReezon, 2022-07-09:
I'm having a similar issue.  I keep having corrupted install warnings.  I have tried installing Octoprint using the Octo4A installer and failed dozens of times across 3 Android devices (2 Samsung phones and 1 LG phone).  I've tried downloading using Wifi only, Carrier data only, Wifi and Carrier data active, and Hotspot data.  I get the same result each time.  Here are some log screenshots that may be helpful in identifying the issue.  After the PEP 517 error, the Octo4A app crashes and the log disappears.  When I restart the app, I see the corrupted install message. 
[image]
[image]
[image]

n3rd253, 2022-07-10:
@feelfreelinux any input here?  I am also getting the exact same crash scenario dispite full permissions granted

therustyrobot, 2022-07-13:
Same issue. Samsung SM-S327VL running Android 6.0.1

xiaotian63, 2022-07-13:
I am getting a few of errors when attempting an install.  The error points seem to vary. It all aligns to one issue possibly. My only guess on cause is permission issues. The app asks for permission, i allow, but i see a small warning “missing storage permission” before and after allowing storage permission. The actual failure points include serialpipe and curl. I don’t know much about coding of linux, so I don’t know what they mean. Reference: I’m using a prepaid phone on wifi only. It is a blu view 2 running android 11. No modifications have been done to the os other than allowing side loading of apps.

xiaotian63, 2022-07-13:
[image]

xiaotian63, 2022-07-13:
> [image] [image]
> 
> Permissions are given, but app crashes when attempting to install. I've tryed a few different versions with no success. Phone is a blue view 2 just out of the box with no other downloads. App crashes after screenshot on the log.

Maybe its android itself. Android 11 did some weird things to storage management and what apps are allowed to do. I bought a blu view 2 too, and it wont work either. I think its android. The only way to get passed this is to have the app permitted full file access, not just file access. I don’t know if that is implementable in a side loaded app tho.

xiaotian63, 2022-07-15:
After a lot of tiral and error, i found version 0.5 works. While not a permanent solution, it is a fast temporary one. Simply back up and restore once 1.1.1 issue gets fixed

tycho94, 2022-07-27:
I had similar issues on an Galaxy S6, I rooted it, installed a custom bootloader and installed android 10, lineage OS 17.1. This solved the issue for me. I had (almost) the same installation issues as @LojikNReezon shows in his screenshots. 

Rooting and such might take some more time but its still better than waiting for a raspi stock :)

cylemmulo-1, 2022-07-29:
> After a lot of tiral and error, i found version 0.5 works. While not a permanent solution, it is a fast temporary one. Simply back up and restore once 1.1.1 issue gets fixed

Yeah this is the only thing that works on my blue view 2. I tried like 4 other higher versions and they all fail. Something changed that I don't think agrees with android 11+

tycho94, 2022-07-29:
> Yeah this is the only thing that works on my blue view 2. I tried like 4 other higher versions and they all fail. Something changed that I don't think agrees with android 11+

I don't think Android 11+ is the issue, I had similar issues with Android 7 until I installed a custom lineage OS with Android 10.

tycho94, 2022-07-29:
This seems to be a reproduction or at least related to #211

alanRfol, 2022-09-11:
I have the same issue with install on Galaxy Note 2 running on O2 network

liamwerk7676, 2022-09-20:
same isue on my nokia 2.4

Bearau, 2022-09-26:
Same issue on Galaxy A51 android 12

KStag13, 2022-09-26:
Same problem with a Galaxy S7 on Android 8

RickzFPV, 2022-12-11:
I Really hope someone can help with getting this going again.  I am like a lot of others that were hoping this was gonna be a alternative to having to spend a ton on Raspberry Pi's.  I run a Print farm with 8 printers and just can't afford to buy PI's for all of them.  I have looked at setting up a linux server to run multiple instances but to this point haven't been sucessful at getting it set up.  I even tried the .5 version talked about above and Can't get a printer to connect through a otg cable.

T4NG0M4N, 2022-12-11:
Yup can confirm, version 0.0.5 installed and works just fine on my Google Pixel 2 running Android 11.

CristianG5401, 2022-12-12:
This fix works for me https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

alufers, 2023-03-24:
I think it should be fixed now.

closed 2023-03-24 by alufers