feelfreelinux/octo4a #238: Application crashed on TV boxes X96max and X96mini

WEREWOLF1973, 2022-07-10:
After installation 1st time octo server is started, can be configured and fully operational.
When i have tryed activate a USB camera, application has crashed. New launch is impossible with same error. 
Rebooting does not help.

Was tested X96max and X96mini with Android 7,9 - same result.

infestius, 2022-07-10:
Hi, same problem on a mecool m8s pro w. You Can try this : go to your Android settings / applications / octoprint / permissions/  here you can revoke camera permission and will disable camera server.
I found that camera server running always made app crash on my Android box.
Feel free to tell if it works too for you

WEREWOLF1973, 2022-07-10:
> Hi, same problem on a mecool m8s pro w. You Can try this : go to your Android settings / applications / octoprint / permissions/ here you can revoke camera permission and will disable camera server. [...]

Thanks, after deleting permission octo severs has started normal way.

alufers, 2022-10-06:
Thanks for the workaround!

alufers, 2022-10-06:
Duplicate of #264

closed 2022-10-06 by alufers