feelfreelinux/octo4a #239: Failed to build tornado

Jonathan1407, 2022-07-11:
Hello folks,
I'm trying to install octo4a on a Samsung galaxy a3 from 2016. Unfortunately, the installation always breaks off at the same point. The message in the log is:
Failed to build tornado

Error: Could not build wheels for tornado which use PEP 517 and cannot be installed directly

Thanks for your Help!

misiektw, 2022-07-24:
I have the same issue on old Tolino Tab 7 (4.2.2). 
Probably "apk add py3-wheels" in install scripts would fix it. Unfortunately I have no idea about Android development toolchain.

But I resolved it by doing Alpine Linux bootstrap manually (or use something like Linux Deploy), then login trough ssh to the Alpine container and install OctoPrint usual way:
```
sudo apk add py3-pip py3-yaml py3-regex py3-netifaces py3-psutil unzip py3-pillow py3-wheel
sudo apk add ttyd ffmpeg python3-dev build-base gcc g++ 
python3 -m pip install OctoPrint
```
Then you can just start Octo as module:
```
python3 -m octoprint serve
```
Still need to figure out how to run this at boot on Alpine.

mantisrobot, 2022-07-29:
I get the same tornado issue on my Samsung SM-T230 running 4.4.2

No idea how to do any of this:

`But I resolved it by doing Alpine Linux bootstrap manually (or use something like Linux Deploy), then login trough ssh to the Alpine container and install OctoPrint usual way:`

so stuck for now, which is a shame as I have a bunch of these tablets doing nothing and printers ready and waiting :)

misiektw, 2022-07-29:
> I get the same tornado issue on my Samsung SM-T230 running 4.4.2
> 
> No idea how to do any of this:

I can give some starting points, but if you don't have working knowledge of Linux, then you will be stuck anyway:

1. Root device
2. Get Linux Deploy APK from https://github.com/meefik/linuxdeploy/releases (probably 2.5.1 because 2.6.0 has API 28 which is way beyond KitKat)
3. Use Linux Deploy to install Alpine Linux with SSHD. 
By this point you have to know at least basics about Linux.
4. Login with ssh to your device (what I used), or use Linux Deploy's build in terminal over http (didn't work for me, but my device is older)
5. Enter commands "apk add.... " i mentioned earlier
6. Try to start module. After that you should have octoprint accessible trough webpage http://your.dev.ice.ip:5000

This is short procedure, but I probably forgetting about some obvious caveats, which of course are obvious to me.
Sorry don't have time to write comprehensive guide, and Octo4a hopefully will fix this soon.
So wait for next release (and pray they will not bump API version to android 5 or 7), or prepare for lots of googling if you're new to Linux.
Good luck ;)

alufers, 2022-10-07:
I think this issue was fixed with the newest version.

Feel free to ping me if it still persists.

closed 2022-10-07 by alufers

sashokbg, 2022-11-30:
Hello @alufers I am using the newest version of octo4a on the exact same devices as the OP but I have the same problem. I hooked my devices to Android Studio for debugging to try understanding what's going on.
Could you please give me some directions ?

The problem happens on  runCommand("cd Octo* && pip3 install .") in OctoPrintHandlerRepository

sashokbg, 2022-12-01:
Hello I have managed to make it work by editing the code and manually installing wheel and tornado with a pip3 command - please follow #319 for more info