feelfreelinux/octo4a #249: Issues with Android 11/12?

cylemmulo-1, 2022-07-28:
Bought a blu view 2 with android 11 on it, and it only gives me the "corrupted installation" error. No sd card in the device. Tried in on my galaxy tab s8 with android 12 and it does the same thing.

jester50158, 2022-08-01:
I have the same issue.  With the same phone. I have tried with an new SD, with out.

cylemmulo-1, 2022-08-01:
> 

It's appearing like a Tracfone and other prepaid vendor issue with some sort of lockdown. I was told to look into boost phones.

Pingnan, 2022-08-15:
Had anyone made any progress on this? I tried anything and still can not install octo4a on my blu view 2.

jup2000, 2022-08-28:
Can confirm 1.1.1 does NOT install on the Tracfone Blu 2.

jup2000, 2022-08-29:
Just purchased a brand new Cricket Moto G Pure running Android 11 with the exact same issue.  Bootstrap problem for ARM installs?

alufers, 2022-10-06:
Hi! Could you post your logs here?

This will help us with debugging.

closed 2023-03-21 by alufers