feelfreelinux/octo4a #250: Installation issue on Samsung MS-T231 tablet running 4.4.2

mantisrobot, 2022-07-29:
When trying t install the latest version it crashes while installing dependencies, so I have tried version 1.0.0 which gets to the end of the install but reports this error at the end of the log:

`could not build wheels for tornado which use PEP 517 and cannot be installed directly`

A google search tells me to update PIP, but have no idea how to do this on a tablet.

karabas2011, 2022-08-14:
I have the same tablet. But I believe it does not have OTG  capability. How do you plan to connect via USB?

mantisrobot, 2022-08-17:
I thought these do have OTG?

alufers, 2022-10-06:
Hi, thanks for trying octo4a!

Please try connecting an USB mouse first. If a coursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

closed 2023-03-24 by alufers