feelfreelinux/octo4a #254: Easy to install, unusable for prints that take more than 2 hours

jonny7737, 2022-08-02:
I have a Nexus 7 2 (2013) that was used for a test twice but is otherwise new (Android 6).  The install was effortless and straight forward.  Setup was just as easy.  Plugged in my OTG adapter and BAM! my printer was on OctoPrint and running great.  Print a little benchy - Flawless :^)

After a bit of fiddling with a new design, I tried to upload gcode to OctoPrint but the OctoPrint server was gone.  Looked at the tablet and the battery had died - with the charger plugged into the OTG adapter.

Five hours of searching and reading I find out that it will never work because the device cannot be OTG and charged at the same time.

What did I miss?

BloomyInDev, 2022-08-05:
Except you test another Y cable or mod your device i think your out of luck

jonny7737, 2022-08-05:
Got it.  A partial solution for very specific, unspecified hardware.

Thanks.

closed 2022-08-05 by jonny7737