feelfreelinux/octo4a #263: Installation error on Tracfone Blu View 2 (Android 11)

TheTrueSquadFan, 2022-08-17:
On verson 1.1.1 whenever I try to download everything from the first app launch, it goes to 100% very quickly then just crashes. I then try the corrupted reinstall option with the same thing. I also tried the latest 2 releases with the same result. Whenever I click the little text log thing, it starts with saying "no bootstrap found" then just goes on. Looking at another post, I tried it with and without a few different vpns on my phone, with nothing changing... anyone have advice?

MelTorme1, 2022-08-18:
Same here. Bootstrap issue. If there's 2 of us here. Got to be more out there. It worked on YouTube. Thanks for this amazing piece of software.

TheTrueSquadFan, 2022-08-18:
> Same here. Bootstrap issue. If there's 2 of us here. Got to be more out there. It worked on YouTube. Thanks for this amazing piece of software.

Yep, gotta be more if the two of us have it around the same time... just gotta hope for a response to get it working

TheTrueSquadFan, 2022-08-18:
I ended up installing 0.0.5 and it worked. Not full time but until this gets fixed... yeah

ndarrow, 2022-08-20:
Same issue here. Blu View 2, 1.1.1 fails during bootstrap.

schuh8, 2022-08-23:
Same as everyone else on Blu View 2 and 1.1.1. But version 0.0.5 installs and appears to work. version 1.1.1 installs in about a minute but shows corrupted when started. Version 0.0.5 takes 10-15 minutes to install but does not indicate corrupted.

smitty-cw, 2022-08-24:
I tried every version from 1.0.0 - 1.1.1 and they all fail to install. Installed 0.0.5 and it works.

junkyardwill, 2022-08-25:
Has anyone figued this out yet? Im trying to see if i cant root my phone and get this to work but so far ive had 0 luck. I have the blu view 2 bl131dl with android 11.

getbiks, 2022-08-27:
I don't think rooting will solve it. I am also having some issues. I tried old version and it runs fine. The newer version has some dependencies which is messing up the installation. You can try installing 0.05 and see if it works for you and use it for time being.

tmoore521, 2022-08-27:
OK ... So I have Octoprint4a  v0.05 working with Tracfone Blu View 2 (Android 11).  I had several network issues with the "New" Elegoo Neptune 3 printer hooking up.  Octoprint does not like all the new claptrap like filament run out etc.  I disabled all that ... Set everything to AUTO and away she went.   UNTIL a new problem reared its ugly head.  The battery on the Blu View 2 is not charging and yes I'm tried an OTG Y cable and also an OTG Dongle.  No luck... Any suggestions or combo of cables that is known to work.  I don't know why this is so difficult the cool kids on YouTube made it work?????

ndarrow, 2022-08-28:
I had three blu view 2 phones working on 0.0.5. I turned them off and back on again, and now they all crash when opening the octo4a app... I feel like this is some kind of scam. After I got the first phone "working" I bought 8 more because I thought this was the solution to the Raspberry Pi shortage. Nope. 
I assume all the YouTubers just get it working for the camera and funnel you to the affiliate links.... I'm sick of the the YouTubers selling out. Off to rage in their comment sections...

ndarrow, 2022-08-28:
Factory reset doesn't fix the problem. When installing ANY octo4a version it crashes. I cannot express how angry I am. Is octo4a supported anymore? I'd pay to get this working.

jup2000, 2022-08-28:
Defintely a Tracfone Blu 2 issue.  Purchased 2 based on YT as well, both broken on install.  They are both going back.

junkyardwill, 2022-08-28:
The blu view 2 was working up until very recently. I’m in a group on Facebook who was recommending this without affiliate links. Sent from my iPhone

tmoore521, 2022-08-28:
OK .... So I'm not quitting on this approach... Octoprint 4a merit .... The battery on the Blu View 2 is not charging and yes I tried an OTG Y cable and also an OTG Dongle / Hub. The print stops until I disconnect the charger. No luck... Any suggestions or combo of cables etc,  that are known to work. 

Thanks Tmo

ndarrow, 2022-08-28:
This worked for me before the app started crashing:

atolla USB C Hub – Aluminum 4-in-1 USB C Adapter with 3 USB 3.0 Ports & 60W Power Delivery Port for MacBook Pro/Air, iPad Pro, Chromebook, Dell and More https://a.co/d/hJEu3Wm

tmoore521, 2022-08-28:
Hmmmm ... That's the one that I'm using  (Atolla USB C Hub – Aluminum 4-in-1 USB C Adapter with 3 USB 3.0 Ports). ...  Can't get the battery to charge. without stopping the print.  Maybe I need a more powerful brick ....  Will check that path. 

Thanks Tmo

junkyardwill, 2022-08-28:
Some manufacturers are disabling dual otg I think. I think they’re trying to deter people from us from buying their equipment to use for other purposes.Sent from my iPhone

tmoore521, 2022-08-29:
Using a higher wattage power brick and voila its charging ... Now let's see if it blows up like what happened to ndarrow. Will post back soom

schuh8, 2022-08-29:
FWIW just too much trouble for what it offers. I gave up and returned the BluView. Now using a ESP32 Cam to monitor print ... no problems at all. Looking to buy a wifi SD card so I can transfer files to printer from another room. Goodby Octopi ... can't say it's been fun.

tmoore521, 2022-08-30:
2 printers running for 2 day .... No new issues yet ... Fingers crossed .... Running the first multi day print ...That should scare up the gremlins.  Will post back in 2 days AND if this works I make a detailed report how I set everything up

tmoore521, 2022-08-31:
Roughly 30 hrs into  the big print marathon .... I had a nozzle clog on the Neptune3 ... Print did not recover due to all the error reporting stuff disabled. Restarted and has run trouble free for  15 hrs or so.  My other printer Monoprice Maker Select is plugging along no issues.  Will RE Post in a day or so.  If there is anyone looking@ this thread Do you want me to post the steps I took to get this system up and running????
NDARROW Would you be so kind and tell what you meant by "I had three blu view 2 phones working on 0.0.5. I turned them off and back on again, and now **they all crash** when opening the octo4a app..

ndarrow, 2022-08-31:
I meant exactly what I said. I had 3 OctoPrint instances working on 3 blu view 2 phones. I shut my farm down for a day so I turned the phones off. When I turned them back on the Octoprint app would crash and close immediately after opening. This was with version 0.0.5. 

Update- I got them working again yesterday. I factory reset all 3 and forced them to install pending updates. They didn’t seem to update properly or there were multiple updates needed because after the first update, it still said updates were available. I updated until it said there were no updates available. Then I left them on, plugged in, and connected to wifi for 24+ hours. Then I installed Octo4a 0.0.5 and they are working again. I haven’t powered them off again because I need them working right now and don’t have time to deal with more nonsense. 

I don’t know what actually made the difference. I updated the phones the first time too. They may still start crashing again if I turn them off. There’s deeper problems that I’m not capable of investigating.

tmoore521, 2022-08-31:
Thanks Ndarrow ... As I suspected the crashes we / are  due to the updates ... I think I have all auto updates off..  I also Removed / Disabled all apps except for necessary google stuff and installed firefox to load octoprint 4a.  I have shutdown and restarted several time ... No crashes so far ... Good Luck

tmoore521, 2022-09-02:
FYI ... Still running trouble free.  
Issue:  During the print marathon Norton 360 on the controlling Win 11 computer decided to do updates and locked Norton up.  The computer stayed on.   I'm in the process of tayloring a new computer to be the printer exclusive rig.  I'll report back with progress updates in a couple of days:
Status:  Both printers running with no errors@ printer (s)

ndarrow, 2022-09-05:
tmoore521, have you turned the phone off and back on again? I think there are multiple issues here- not just from auto updates. When the phone is shut off and turned on again, the octo4a apps crash.

I have experienced the same problem with 5 phones multiple times. 

- I got them all working on octo4a v 0.0.5 (installed all of the system updates, let the phones settle for many hours to let background updates finish, installed octo4a, connected printer, got octoprint setup, completed several 12 hour+ prints...)
- Then I shut down the print farm (turned off the breakers to the print farm- which includes the phones' power supplies. I have usb power blockers between the usb hub and the printer) 
- The phones' batteries drain over time and the phones shut off. 
- Now I powered the farm back up and all 5 apps crash as soon as I try to open them... 

So, unless you can keep them powered 24/7 (printing or not) the app will become unusable and require several hours of time to factory reset, install updates, wait for background updates, setup router settings/ip's/device names, reinstall octo4a, setup octoprint...  It's fine until they power off. Then you're screwed.

ndarrow, 2022-09-05:
I jumped the gun apparently. 

While messing with settings on one phone, it started working without having to factory reset or reinstall octo4a. I assumed it was some random developer setting I messed with so I went to compare to one of the other phones I hadn't monkeyed with and that phone's app just started working out of nowhere (didn't change any settings). The rest of the phone started working too. 
I did nothing and they went from: app crashing immediately, to working perfectly. 

Time was the only difference (about an hour).
I suspect it's either a minimum battery % that needs achieved before it will allow the app to run, or after the phone powers on it needs about an hour for background startup junk to settle. 

overall, I'm still not convinced this is reliable, but happy that I didn't have to factory reset all of them- this time... 

I realize my particular issue isn't exactly the same as the title of the issue (sorry for spamming everyone), but others will likely have the same issues as I am if they are reading this particular issue. 
It would be great if the next release supported the Blu View 2. I suspect the issues I've experienced would improve at the same time.

cdunn95, 2022-09-10:
@ndarrow @schuh8 @tmoore521 

Hi!

I'm a mobile software engineer that just happened to stumble across this issue while I was attempting to set up an old nexus 7 tablet I had lying around.

The issue you are experiencing is not a product of BLU trying to put a damper in your phone, it is one of the releases. 

I have tinkered the hell out of my Nexus 7 tonight (poor thing has probably gone back and forth between Android 4.4 and 7.1.2 two-dozen times at this point), and have found that Android 4.4 (KitKat) WILL install the application correctly and allow for the server to launch – unfortunately for me, and luckily for this community, KitKat does not allow for charge over OTG without custom kernels that all have broken links. 

Anyways, after reviewing your comments and seeing that 0.5 is working for some, I did some digging in the comparison between that release and the current 1.1.1 and I think I may have narrowed down a culprit. 

Inside of the BootstrapRepository.kt file, the author has appended this line in the newer releases: 

```private fun shouldUsePre5Bootstrap(): Boolean {
        if (getArchString() != "arm" && getArchString() != "i686") {
            return false
        }

        return Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP
    }
```

Which is then utilized later on in the same file as seen in the edited code here:

```
if (shouldUsePre5Bootstrap()) {
                    runCommand("rm -r root && mv root-pre5 root", prooted = false).waitAndPrintOutput()
                }
```

This seems to be the culprit as, as many have stated, when on a newer version of Android the output is almost instantaneous, I.E it is not "waiting" and "printing output". This could* make this into a race condition situation where a dependency that is required to install the following extensions don't have that, and thus... corrupt install.  

I have not dived deeper into this because... well... It's currently 9:45AM on a Saturday and I just pulled an all-nighter in an attempt to resolve this for myself.

Sometime this weekend (once I have repaid my sleep debt) I will look more into this and see if I can't come up with a working resolution, at which point I will update this ticket and release a pull request to the author. 

Disclosure: Although I am in mobile software engineering, I am primarily an iOS developer, so I'm not super familiar with Android Studio; as such it may take me a bit before I can get an actual APK out to anyone while I familiarize myself with the IDE/packaging process. Also, if possible I would like to allow the original author a chance to release it on his own terms out of respect for the project. 

But... for now.... sleepy time. 

P.s. Go buy your phones back lol, they didn't do anything.

tmoore521, 2022-09-10:
All,
The way I have everything configured (cobbled) is working.  I can cycle off and on, (with any amount of time between cycle) and everything wakes up and works just like the time before.  Lucky I guess realizing that my efforts  are a bandaid fix!  
Cameron,  Thanks for working this issue!  Looking forward to you deep dive solving the issue.  I know of at least 3 people that will be VERY happy.

Tom

ndarrow, 2022-09-15:
@cdunn95 
I really appreciate your support here. I can't understate how badly I need this to work. 

General update:
I now have one phone that won't boot Octoprint for the first time. It just sits there spinning... I left it for 24 hours and it never booted. I've cleared cache, cleared data, uninstalled/reinstalled (0.0.5), and factory reset with no success. Originally, the app started crashing after it was turned off/on, so I factory reset, updated, let it sit for 24 hours to settle, and when I installed 0.0.5 it won't boot Octoprint. 

The other phones seem stable enough, as long as I never let them turn off.

zapdawg, 2022-10-01:
I just purchased 2 phones Oct. 1 2022. One it turns out is a BLB131DL and it is having the same problem.  The other is a BLB130DL and it is working fine.  The first is android 11 the BLB130DL is android 10. I have another 130 and it has been working for months. I going back to the store to see if there is another 130 to swap for until this problem has been fixed.

zapdawg, 2022-10-01:
the back of the box says Learn more about this phone and gives the model number.

zapdawg, 2022-10-01:
the bottom of the box tells you the model too

ndarrow, 2022-10-12:
Update- All 5 instances I had working have permanently stopped. No amount of updating, factory resetting, or uninstalling fixes the problem.  When attempting to open the app, it just crashes immediately. The creator should really just delete octo4a so more people aren't lured into this situation...

gbw80521, 2022-10-18:
> Update- All 5 instances I had working have permanently stopped. No amount of updating, factory resetting, or uninstalling fixes the problem. When attempting to open the app, it just crashes immediately. The creator should really just delete octo4a so more people aren't lured into this situation...

@ndarrow - Just wanted to let you know that I have been following this thread since I have been having the same problems with o4a not working on a Blu View 2. I noticed today that Octoprint released a new version and thought I would give installing another shot and 0.0.5 WORKED and successfully installed and the server booted (where it always hung for me before). Current release still fails the same way as it was before but if 0.0.5 works, at least that is something!

Just wanted to give you a heads up so maybe you could try redoing those installs on your phones before something changes again and causes issues.

Hope it works for you too!

ndarrow, 2022-12-09:
> So just curious has anyone been able to find out if this issue is a Android 11 problem or Octo4A problem I just recently bought a Blu View 2 that updated to Android 11 last night and I am having the same issue with trying to install octo 4a just wondering if I should take this phone back and get something different. I have it running perfectly on my lenovo Tablet on Android 10 and It works perfectly

Take it back and forget octo4a exists. I’ve resorted to mining raspberry pi’s from various Micro Centers around the country using a network of friends that live “near” them. They’re popping up in stock at MCs every so often. 

Or, try reverting to Android 10 and see if you gain anything.

tmoore521, 2022-12-09:
The blue View 2 can be successfully set up … It’s Tricky … Use Octo4a ver 0.5.  B4 you load octo 4a strip the phone to it’s basic operating by uninstalling all apps and disable the ones that won’t uninstall.  The exceptions are  l1. Load Fire Fox as your browser.. 2. Do not disable anything related to the google base operations    3. Load Octo4a thru Github only

alufers, 2023-04-01:
It should work in 1.2.4.

closed 2023-04-01 by alufers