feelfreelinux/octo4a #264: App crash on startup

metalichesky, 2022-08-19:
After installing klipper using sh script over ssh connection, I kill app and restart it from launcher, then app crash happens. Looks like octo4a CameraService tries to select camera, but my device has no camera at all, so list of cameras is null. How camera service can be disabled in that case?

Device: Ugoose X3 Pro
Android Version: 9
App version: octo4a 1.1.1

```
12899-12899/com.octo4a E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.octo4a, PID: 12899
    java.lang.NullPointerException
        at com.octo4a.camera.CameraService$cameraSelector$2.invoke(CameraService.kt:67)
        at com.octo4a.camera.CameraService$cameraSelector$2.invoke(CameraService.kt:36)
        at kotlin.SynchronizedLazyImpl.getValue(LazyJVM.kt:74)
        at com.octo4a.camera.CameraService.getCameraSelector(Unknown Source:2)
        at com.octo4a.camera.CameraService.access$getCameraSelector$p(CameraService.kt:36)
        at com.octo4a.camera.CameraService$readyCamera$1.run(CameraService.kt:265)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6669)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

metalichesky, 2022-08-19:
If someone encounters this problem, it is enough to revoke the granted permissions to use the camera in system settings for application. It seems that if the application does not have permission, then it does not try to look for the camera.

Thebigskysky, 2022-09-15:
> If someone encounters this problem, it is enough to revoke the granted permissions to use the camera in system settings for application. It seems that if the application does not have permission, then it does not try to look for the camera.

You sir are a saint... been beating myself up over trying to keep octoprint running on my T95 box...

commit 85ff29b3 (2022-10-07): fix: app crash when running on device with not cameras and having camera permission

fixes: #264