feelfreelinux/octo4a #271: Unable to Install / Corrupt Install Android 7.0 Samsung Note 5

MidTN3dRC, 2022-09-06:
Attempted to install 1.1.1 multiple times 
Stepped back to a previous apk @ 1.0.3
Both failed and crashed the Octoprint For Android app.

Log and screenshots are as follows;

[image]
[image]

MidTN3dRC, 2022-09-06:
inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/htmlfile.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/jsonp.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/pollingbase.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/streamingbase.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/websocket.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/xhr.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/transports/xhrstreaming.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/util.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint/vendor/sockjs/tornado/websocket.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/src/octoprint_client/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint_client/__init__.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/src/octoprint_setuptools/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/src/octoprint_setuptools/__init__.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/access/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/__init__.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/access/_plugins/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/_plugins/permissions_plugin.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/access/groups/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/groups/__init__.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/groups/test_groupmanager.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/test_permissions.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/access/users/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/users/__init__.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/users/test_usermanager.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/access/users/test_users.py   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/cypress.json   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/fixtures/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/fixtures/e2e-test.gcode   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/fixtures/example.json   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸    creating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/connect.js   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/login.js   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/no_errors.js   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/settings.js   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/integration/upload.js   (warning) cannot set modif./access times
🐸           Operation not permitted
🐸   inflating: OctoPrint-OctoPrint-87f4a4e/tests/cypress/package-lock.json   (warning) cannot set modif./access times
🐸           Operation not permitted
[... cut ...]
🐸 Using legacy 'setup.py install' for zipstream-ng, since package 'wheel' is not installed.
🐸 Using legacy 'setup.py install' for sgmllib3k, since package 'wheel' is not installed.
🐸 Building wheels for collected packages: tornado
🐸   Building wheel for tornado (PEP 517): started
🐸   Building wheel for tornado (PEP 517): finished with status 'done'
🐸   WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted
🐸 Failed to build tornado
🐸 ERROR: Could not build wheels for tornado which use PEP 517 and cannot be installed directly
💻 Dependencies installed

esanchez72223, 2022-09-15:
Hey, I am curious if you were able to resolve this issue. I am having the same exact problem, same phone, same android version. Cant seem to figure it out.

MidTN3dRC, 2022-09-15:
> Hey, I am curious if you were able to resolve this issue. I am having the same exact problem, same phone, same android version. Cant seem to figure it out.

No I have not been able to resolve it. 
I will likely move on and use an alternative for Octoprint. I was able to get it working on my newer Samsung S22+ daily use phone so I will just try with some other older android devices I have available and find another use for the Note 5.

esanchez72223, 2022-09-15:
Funny, I actually did the same thing with my Samsung S20 FE and loaded it fine as well. I was doing some research and found that apparently people have had success rooting their old android phone and installing android 10, lineage OS 17.1. I'm savvy, but I am by no means and computer science guy. But I did find some instructions and videos online of how to do this and will be giving it a shot. I'll probably give it a shot this weekend and comment back to let you know if I was able to get it to work.

KStag13, 2022-09-26:
I'm having a very similar issue on a Samsung Galaxy S7, Android 8. No matter how I give it storage permissions, it pops up "missing storage permissions" and then displays "Install Corrupt" at the top of the options.

claudiubalogh, 2022-10-03:
Same problem for a Galaxy S5 Duos (SM-G900FD) with Android 6.0.1. Side note: octo4a would be a perfect use for this phone, since it has a broken screen (not cracked, just does not display anything). Thank you!

loxire, 2022-10-15:
Having the same issue on my galaxy j5 and s7

ThePunisher, 2022-11-07:
Same on S4 19505, Android 5.0.1 Lollipop, no SD-Card installed.

sashokbg, 2022-11-29:
Same issue here with Galaxy A3 2016. Is it possible to grant root privileges to the app for those who have rooted androids ?