feelfreelinux/octo4a #272: Not connecting to ender 3

Zoomy51, 2022-09-10:
I cannot get my phone to connect to my ender 3 original. I am using an old LG phone running android 5 lollipop, and I have also tried my Pixel 3 running android 12. I am using a USB hub that has been proven to transfer data. I don't know what the issue is.

bjohnrini, 2022-09-29:
Any luck?

kadamski, 2022-10-06:
@Zoomy51 have you verified that the OTG works on your phone(s) at all? There are also several Ender 3 boards, they may have a different USB->serial chip so I guess it would make sense to verify which one you have. Showing the output of lsusb when then it is connected to some other computer would show this. Maybe your chip is not on the list (https://github.com/feelfreelinux/octo4a/blob/master/app/app/src/main/res/xml/device_filter.xml)?

alufers, 2022-10-06:
Hi! Please try connecting an USB mouse first. If a coursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.
Hi, thanks for trying octo4a!

Please try connecting an USB mouse first. If a coursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

Zoomy51, 2022-10-08:
I tried a USB mouse on both my LG-90 and my Pixel 3. No cursor appears. I also tried without my OTG cable, and it didn't work. Did I get unlucky with 2 phones that don't support it or do I need a different OTG cable? What cables worked for you?

bjohnrini, 2022-10-08:
I was able to get my moto x4 working using https://www.amazon.com/gp/product/B08X4LHP1B/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
The cable from the hub needs to be attached to my phone. The printer needs to be turned ON with everything plugged in first.

Zoomy51, 2022-10-09:
Ok. I was using a USB hub by Anker but I just bought an OTG hub like the ones in the pictures. I'll update you guys when it arrives.

Zoomy51, 2022-10-10:
Ok guys. The OTG arrived. I've tested 3 phones, 2 of which didn't detect the printer, so I'm going to eliminate them from this. The third one however, did connect to the printer, but when trying to install the octoprint files from the app, the app crashes. I can't figure out why. I'm running octo4a 1.1.2. I also tried 1.1.1