feelfreelinux/octo4a #278: No usb popup on Xperia L1

jrbanach, 2022-09-19:
Octoprint seemed to install fine, but when i tried to plug in the printer to the app, i never get any popup about the usb device.  I tried a few different OTG/HUB devices and also a few usb cables (including taping the power supply pin), it never shows up even with only the power usb cable disconnected.

the device shows up fine on my pc (Ender 3v2 ch340).

i tried reinstalling the octo4a app on the phone, but that didn't seem to help.  I cant find anywhere that the USB module installed (or didn't install successfully) and i didn't see an option to manually grant it permissions either in android settings.

thoughts?

alufers, 2022-10-06:
Hi, thanks for trying Octo4a!

Please try connecting an USB mouse first. If a cursor doesn't appear this means this is a problem with the OTG cable or your phone does not support OTG.

Cheers!

closed 2022-10-06 by alufers

rafatozzi, 2023-05-21:
I have the same problem and did the same test.

I put the mouse on device, the cursor appears normally