feelfreelinux/octo4a #281: Won't install Octoprint 1.83

rfnovo, 2022-09-21:
I tried to update my Ocroptint install for 1.83 and received several errors during upgrade.

Decided to reinstall Octo4a from scratch to clean up my Octoprint install, but now I'm receiving the same error during Ocroptint setup.

Can't capture screen nor the log since it will crash Octo4A when Octoprint install fails

degroof, 2022-09-21:
I had the same problem. Trying to install on Pixel 2. Here's the log from mine...

[failed-183.log](https://github.com/feelfreelinux/octo4a/files/9617131/failed-183.log)

ltsharma, 2022-09-22:
Same here as well, tried in 2 phones.
Mi 4i (Android 5)
Mi K20 Pro (Android 11)

degroof, 2022-09-22:
Looks like this pull request is intended to fix the issue.
https://github.com/feelfreelinux/octo4a/pull/280

pnmartinez, 2022-09-22:
Same here.
Xperia Z3, Android 5.

kadzur, 2022-09-22:
Ditto Samsung Galaxy S9+, Android 11

Madcamper, 2022-09-22:
The workaround cp2004 performed was on a fresh install. If you are trying the workaround from 1.8.2 > 1.8.3, it won't work. When I performed the workaround, it downgraded me from 1.8.2 to 1.8.1 and still can't update to 1.8.3. Looks like will have to wait for the maintainer to push an update.

feelfreelinux, 2022-09-22:
Thanks for the workaround! Looking into the issue :)

w7boz, 2022-09-23:
Will we be notified when this is fixed?

Kevin

rfnovo, 2022-09-23:
> Will we be notified when this is fixed?
> 
> Kevin

just set to watch the repository and you will be notified when a update is released

feelfreelinux, 2022-09-27:
Fixed in https://github.com/feelfreelinux/octo4a/releases/tag/1.1.2 :)

closed 2022-09-27 by feelfreelinux