feelfreelinux/octo4a #282: Octo4a Never installs

w7boz, 2022-09-22:


w7boz, 2022-09-22:
Installation Corrupted
[octo4a.txt](https://github.com/feelfreelinux/octo4a/files/9625628/octo4a.txt)

rfnovo, 2022-09-22:
Same issue as https://github.com/feelfreelinux/octo4a/issues/281

pnmartinez, 2022-09-22:
Same here.
Xperia Z3, Android 5.

lauripaulo, 2022-09-23:
I seems to be a problem with Octoprint 1.8.3 dependencies. I am having the same issue. I was able to overcome this and install octoprint by fixing the version to 1.8.1 in the source code then compiling with Android Studio. Yes, dirty and ugly fix but at least I found the problem. 

I also harcoded the apk dependencies there while trying to make 1.8.3 works.

Hope it helps a proper fix in the code.

[image]

rfnovo, 2022-09-23:
> I seems to be a problem with Octoprint 1.8.3 dependencies. I am having the same issue. I was able to overcome this and install octoprint by fixing the version to 1.8.1 in the source code then [...]

Mind share the APK?

lauripaulo, 2022-09-23:
> Mind share the APK?

I just compiled for me a "app-debug.apk" not a final or proper version. It's better to wait for the oficial fix. If you really want to test it, with all the problems it might have, let me know.

w7boz, 2022-09-24:
I would like to try it.

pnmartinez, 2022-09-24:
> I just compiled for me a "app-debug.apk" not a final or proper version. It's better to wait for the oficial fix. If you really want to test it, with all the problems it might have, let me know.

Yeah! Another one interested on testing your APK here!

fbreda, 2022-09-24:
I would like to try it.

> I just compiled for me a "app-debug.apk" not a final or proper version. It's better to wait for the oficial fix. If you really want to test it, with all the problems it might have, let me know.

I would like to try it, too

protocold, 2022-09-25:
ya could you please post your apk?

lauripaulo, 2022-09-25:
You can get an alpha release here:

https://github.com/lauripaulo/octo4a/releases/tag/octo4a-debug-fixed-octoprint-1.8.1

Please, use at your own risk. Upgrade to the official version when the problem is solved.

ChrisIsCross, 2022-09-25:
so i just installed your version and it works for the first time! thanks for the work!

pnmartinez, 2022-09-26:
> You can get an alpha release here:
> 
> https://github.com/lauripaulo/octo4a/releases/tag/octo4a-debug-fixed-octoprint-1.8.1
> 
> Please, use at your own risk. Upgrade to the official version when the problem is solved.

Here installed correctly too!

Only thing that crashes yet is **webcam support**, specifically when I try to set `Camera source` or `Camera resolution`.

ss7admin, 2022-09-27:
Thanks a lot. It worked

w7boz, 2022-09-27:
I just installed it and it worked! Ok, kind of... I enabled the Camera and it crashed. So I tried it again, and this time the Camera is working. Running a print right now. But can't wait to hook it up to the printer! I really love this APP!!!!

Kevin

feelfreelinux, 2022-09-27:
Fixed in https://github.com/feelfreelinux/octo4a/releases/tag/1.1.2 :)

closed 2022-09-27 by feelfreelinux