feelfreelinux/octo4a #284: Crash in Install v1.1.1 (and older)

darkza, 2022-09-25:
Deleted Log a bit because to much Charakters. >
[image]

🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/test_daemon.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/test_events.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/test_octoprint_setuptools.py 
🐸    creating: OctoPrint-OctoPrint-d0d226f/tests/timelapse/
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/timelapse/__init__.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/timelapse/test_timelapse_helpers.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/timelapse/test_timelapse_renderjob.py 
🐸    creating: OctoPrint-OctoPrint-d0d226f/tests/util/
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/__init__.py 
🐸    creating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/
🐸    creating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/jinja_test_data/
🐸  extracting: OctoPrint-OctoPrint-d0d226f/tests/util/_files/jinja_test_data/.hidden_everywhere.txt 
🐸  extracting: OctoPrint-OctoPrint-d0d226f/tests/util/_files/jinja_test_data/normal_text.txt 
🐸  extracting: OctoPrint-OctoPrint-d0d226f/tests/util/_files/jinja_test_data/not_a_text.dat 
🐸    creating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/already_installed_1.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/already_installed_2.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/egg_problem_1.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/egg_problem_2.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/python_mismatch_1.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/python_mismatch_2.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/pip_test_data/successful_install_1.txt 
🐸  extracting: OctoPrint-OctoPrint-d0d226f/tests/util/_files/utf8_with_bom.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/_files/utf8_without_bom.txt 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_caseinsensitive_set.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_comm.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_comm_helpers.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_commandline.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_counted_event.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_fast_deepcopy.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_file_helpers.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_files.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_jinja.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_json.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_misc.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_net.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_pip.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_platform.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_repeated_timer.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_resettable_timer.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_string_helpers.py 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/tests/util/test_version.py 
🐸    creating: OctoPrint-OctoPrint-d0d226f/translations/
🐸    creating: OctoPrint-OctoPrint-d0d226f/translations/de/
🐸    creating: OctoPrint-OctoPrint-d0d226f/translations/de/LC_MESSAGES/
🐸   inflating: OctoPrint-OctoPrint-d0d226f/translations/de/LC_MESSAGES/messages.mo 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/translations/de/LC_MESSAGES/messages.po 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/translations/messages.pot 
🐸   inflating: OctoPrint-OctoPrint-d0d226f/versioneer.py 
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Processing /root/OctoPrint-OctoPrint-d0d226f
🐸 Collecting OctoPrint-FileCheck>=2021.2.23
🐸   Downloading OctoPrint_FileCheck-2021.2.23-py2.py3-none-any.whl (19 kB)
🐸 Collecting OctoPrint-FirmwareCheck>=2021.10.11
🐸   Downloading OctoPrint_FirmwareCheck-2021.10.11-py2.py3-none-any.whl (33 kB)
🐸 Collecting OctoPrint-PiSupport>=2022.3.28
🐸   Downloading OctoPrint_PiSupport-2022.6.13-py2.py3-none-any.whl (33 kB)
🐸 Collecting argon2_cffi<22,>=21.3.0
🐸   Downloading argon2_cffi-21.3.0-py3-none-any.whl (14 kB)
🐸 Collecting cachelib<0.3,>=0.2
🐸   Downloading cachelib-0.2.0-py3-none-any.whl (12 kB)
🐸 Collecting Click<9,>=8.0.3
🐸   Downloading click-8.1.3-py3-none-any.whl (96 kB)
🐸 Collecting colorlog<7,>=6
🐸   Downloading colorlog-6.7.0-py2.py3-none-any.whl (11 kB)
🐸 Collecting emoji<2,>=1.4.2
🐸   Downloading emoji-1.7.0.tar.gz (175 kB)
[... 211 lines cut ...]
🐸       copying cffi/cparser.py -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/verifier.py -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/lock.py -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/pkgconfig.py -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/_cffi_include.h -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/parse_c_type.h -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/_embedding.h -> build/lib.linux-aarch64-3.9/cffi
🐸       copying cffi/_cffi_errors.h -> build/lib.linux-aarch64-3.9/cffi
🐸       running build_ext
🐸       building '_cffi_backend' extension
🐸       creating build/temp.linux-aarch64-3.9
🐸       creating build/temp.linux-aarch64-3.9/c
🐸       gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DFFI_BUILDING=1 -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.9 -c c/_cffi_backend.c -o build/temp.linux-aarch64-3.9/c/_cffi_backend.o
🐸       error: command 'gcc' failed: No such file or directory
🐸       ----------------------------------------
🐸   ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cd68c4ec/cffi_923aa3a2bf1c44c6855fc0472f92748f/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cd68c4ec/cffi_923aa3a2bf1c44c6855fc0472f92748f/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-qhr_ucpc/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-9ujkcvc3/overlay --compile --install-headers /tmp/pip-build-env-9ujkcvc3/overlay/include/python3.9/cffi Check the logs for full command output.
🐸   ----------------------------------------
🐸 WARNING: Discarding https://files.pythonhosted.org/packages/fe/c7/1796c08bc2834424326bec6fffb81bd082cb6c78b278fd0a55b2a09b0454/argon2-cffi-bindings-21.1.0.tar.gz#sha256=5b3c72b092761e9aa8a8335096c409e9d68d127ae64520848d1d9d47459c4650 (from https://pypi.org/simple/argon2-cffi-bindings/) (requires-python:>=3.6). Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3.9/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-9ujkcvc3/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools>=45' 'setuptools_scm>=6.2' wheel 'cffi>=1.0' Check the logs for full command output.
🐸 INFO: pip is looking at multiple versions of sarge to determine which version is compatible with other requirements. This could take a while.
🐸 INFO: pip is looking at multiple versions of <Python from Requires-Python> to determine which version is compatible with other requirements. This could take a while.
🐸 INFO: pip is looking at multiple versions of octoprint to determine which version is compatible with other requirements. This could take a while.
🐸 ERROR: Could not find a version that satisfies the requirement argon2-cffi-bindings (from argon2-cffi)
🐸 ERROR: No matching distribution found for argon2-cffi-bindings
 
[image]

pnmartinez, 2022-09-25:
Same issue as #282

feelfreelinux, 2022-09-27:
Fixed in https://github.com/feelfreelinux/octo4a/releases/tag/1.1.2 :)

closed 2022-09-27 by feelfreelinux