feelfreelinux/octo4a #287: printer cannot be detected with OTG via USB powered HUB

protocold, 2022-09-27:
I have an old LG Optimus G phone. The only way to get it to do OTG is to connect via a powered hub, which I confirmed to work with other USB devices including: mouse, keyboard and USB storage. All of them got detected without issue.

However, after octo4a is successfully installed and started, I plugged in the printer but it couldnt detect the printer, I have also went to the web gui and click on connect and can see the log on the phone that it has attempted to probe for several things like different speed etc but none of that worked... 

Is there a way to troubleshoot this? Does it has to do with the way I hook up the device (via a USB hub)?

protocold, 2022-09-30:
the line last -> Bus 002 Device 005: ID 1a86:7523 is the printer but it doesnt show up in Octoprint/Octo4a... how can I make this work?

alufers, 2022-10-06:
Is this lsusb otput from your android phone or a PC?

protocold, 2022-10-07:
android phone.

protocold, 2023-06-03:
hi there... any possible update on this? do you have a pointer if i want to try to take a look at this?