feelfreelinux/octo4a #288: Ver 1.8.4 not installing

degroof, 2022-09-27:
Seems to bet failing at this point:

```
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DFFI_BUILDING=1 -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.9 -c c/_cffi_backend.c -o build/temp.linux-aarch64-3.9/c/_cffi_backend.o
c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
15 | #include <ffi.h>
[failed-184.log](https://github.com/feelfreelinux/octo4a/files/9660061/failed-184.log)

|          ^~~~~~~
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1
```

Log attached.

[failed-184.log](https://github.com/feelfreelinux/octo4a/files/9660062/failed-184.log)

degroof, 2022-09-27:
Sorry. Ignore this. I'm an idiot. Didn't realize I had to install the new octo4a apk _before_ updating Octoprint.

feelfreelinux, 2022-09-27:
Glad to hear so :)

closed 2022-09-27 by feelfreelinux