feelfreelinux/octo4a #289: OTG Y Cable not charging and connected to printer at the same time

ChrisIsCross, 2022-09-28:
I only can connect to the printer and let it run on my phone battery , or i can charge my phone. Not both at the same time. I have tried 2 different OTG Y cables with my Honor 8a and its not woring. Is it the phone or am i doing someting wrong? My printer is the Ender 3 V2. 

Thanks!

alufers, 2022-10-06:
Hi, Thanks for trying octo4a!
Unfortunately some phones just don't support charging and using OTG at the same time :/
This is a known problem, and unfrotunately it is near-impossible to solve on many phones.

We maintain a crowdsourced compatibility list of phones which work with octo4a:
The list: https://docs.google.com/spreadsheets/d/1pbKAELzvTrMVm6NpDpK8TfQR33w3QHQEgqkcGdbPFI0/edit?usp=sharing
Feel free to submit your phone here: https://forms.gle/UFzJebj67hCtLCBz7

Sometimes it is possible to bypass this problem by trying theese solutions:

- Using wirless charging. If your phone supports it, it might be possible to use a wireless charger and use the OTG feature at the same time. Some users have resorted to gluing a wirless charger to a phone case
  [image]
  
- Hard-wiring the battery. 
  **WARNING: CONNECTING A RANDOM POWER-SUPPLY STRAIGHT TO THE BATTERY MIGHT RESULT IN A FIRE OR EXPLOSION, EXERCISE EXTREME CAUTION** 
   Some people modify their phone to charge their battery externally, bypassing the phone's battery charging circutry alltogether. Some event removed the battery (the safest option)
   [image]
- Compiling a non-standard kernel. On some models the charing-and-OTG functionality is blocked in software, and this can be bypassed by using a non-stock kernel. Please be advised that compiling an Android kernel requires some skill and a lot of time.

If you need some guidance please feel free to ask on our telegram group: (https://t.me/octo4achat)[https://t.me/octo4achat]. The community is very likely to help you.

closed 2022-10-06 by alufers

BacaR00T, 2023-10-24:
Same with my Redmi 4A, but the Redmi 6A works just fine