feelfreelinux/octo4a #293: USB serial driver for Malyan 3D printer (Monoprice Select Mini V2)

brianreese, 2022-10-03:
My printer shows up in octo4a as "Malyan 3D Printer", but octo4a doesn't automatically select a serial driver. I've manually tried them all at this point, but have been unable to connect to the printer in octoprint. I've also installed the octoprint plugin "Malyan/Monoprice Connection Fix" - this was helpful when I had the printer connected to my raspberry pi running octopi.

Any idea which driver I should be using? Or how I might begin to debug this issue further?

Octoprint terminal output from the octo4a octoprint instance when attempting a connection (using the CH341 driver):
```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to: /dev/ttyOcto4a
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x7ba21b5610, open=True>(port='/dev/ttyOcto4a', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

I don't know if there's anything useful here, but for comparison, here's the output from my raspberry pi:
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
Trying port /dev/ttyACM0, baudrate 115200
Connecting to: /dev/ttyACM0
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xa3a58c30, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: ok N0 P15 B15
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: ok N0 P15 B15
Send: N1 M115*39
Recv: NAME. Malyan	VER: 4.1	MODEL: M200	HW: HH02
Recv: CLK:EXT BUILD: Jan 24 2018 09:31:06
Recv: ok N1 P15 B15
Send: N2 M21*18
Recv: ok N2 P15 B15
Send: N3 M21*19
Recv: ok N3 P15 B15
Send: N4 M20*21
Recv: Begin file list
Recv: arm.gcode
Recv: MSMV2E3D_Lid_Without_Holes.gcode
Recv: MSMV2E3D_Dispenser_Module_Internal.gcode
Recv: End file list
Recv: ok N4 P15 B15
Send: N5 M105*34
Recv: ok N5 P15 B15 T:17.1 /0.0 B:18.3 /0.0 T0:17.1 /0.0 @:0 B@:0
Send: N6 M105*33
Recv: ok N6 P15 B15 T:17.2 /0.0 B:22.0 /0.0 T0:17.2 /0.0 @:0 B@:0
Send: N7 M105*32
Recv: ok N7 P15 B15 T:18.8 /0.0 B:17.7 /0.0 T0:18.8 /0.0 @:0 B@:0
Send: N8 M105*47
Recv: ok N8 P15 B15 T:19.2 /0.0 B:24.3 /0.0 T0:19.2 /0.0 @:0 B@:0
Send: N9 M105*46
Recv: ok N9 P15 B15 T:21.9 /0.0 B:20.0 /0.0 T0:21.9 /0.0 @:0 B@:0
Send: N10 M105*22
Recv: ok N10 P15 B15 T:22.7 /0.0 B:18.1 /0.0 T0:22.7 /0.0 @:0 B@:0
Send: N11 M105*23
Recv: ok N11 P15 B15 T:15.8 /0.0 B:19.0 /0.0 T0:15.8 /0.0 @:0 B@:0
Send: N12 M105*20
Recv: ok N12 P15 B15 T:22.5 /0.0 B:18.1 /0.0 T0:22.5 /0.0 @:0 B@:0
Send: N13 M105*21
Recv: ok N13 P15 B15 T:23.8 /0.0 B:15.6 /0.0 T0:23.8 /0.0 @:0 B@:0
Send: N14 M105*18
Recv: ok N14 P15 B15 T:16.1 /0.0 B:20.8 /0.0 T0:16.1 /0.0 @:0 B@:0
Send: N15 M105*19
Recv: ok N15 P15 B15 T:15.8 /0.0 B:19.6 /0.0 T0:15.8 /0.0 @:0 B@:0
Send: N16 M105*16
Recv: ok N16 P15 B15 T:17.0 /0.0 B:21.6 /0.0 T0:17.0 /0.0 @:0 B@:0
Send: N17 M105*17
Recv: ok N17 P15 B15 T:23.1 /0.0 B:15.8 /0.0 T0:23.1 /0.0 @:0 B@:0
Send: N18 M105*30
Recv: ok N18 P15 B15 T:18.4 /0.0 B:23.7 /0.0 T0:18.4 /0.0 @:0 B@:0
Send: N19 M105*31
Recv: ok N19 P15 B15 T:21.3 /0.0 B:20.6 /0.0 T0:21.3 /0.0 @:0 B@:0
Send: N20 M105*21
Recv: ok N20 P15 B15 T:15.8 /0.0 B:20.2 /0.0 T0:15.8 /0.0 @:0 B@:0
Send: N21 M105*20
Recv: ok N21 P15 B15 T:20.0 /0.0 B:17.1 /0.0 T0:20.0 /0.0 @:0 B@:0
```

ben-carson, 2022-10-17:
Having the exact same issue on the same printer.

MattLD13, 2022-10-22:
I am also having this issue

GoddessRandi, 2023-05-08:
I was able to get it to connect from a couple different phones by manually changing serial driver in the app and selecting CDC/ACM.

Hope this helps!

D5KDeutsche, 2023-05-08:
I (sort of) wish I could test this. I ended up replacing the mainboard with and SKR. This, among many other problems, are gone.

CubE135, 2024-02-12:
I have the exact same problem. Also using a Malyan M200... found no solution yet :(

kamocat, 2024-04-03:
I also could not start communication with the 3D printer.
I had no trouble using octoprint on my Linux laptop. `lsusb -t` says the driver is cdc_acm, which is expected since it shows up as `/dev/ttyACM0`.

Interestingly, I can plug in a FTDI232R, select the FTDI driver, and get the same result: Fail to open connection. Based on the LEDs, it looks like it didn't even send a single UART frame.

The Malyan/Monoprice plugin doesn't **fix** the connection, but it does make it fail faster. My guess is it's a permissions issue.