feelfreelinux/octo4a #294: Unable to connect to ender 3 v2 with 4.2.7 motherboard

Alex-Bellerive, 2022-10-03:
Recently installed octo4a on an old lg phone running android 5.1.1 and am unable to connect it to my ender 3v2 with the 4.2.7 motherboard. I have verified that the cables and otg spitter work correctly but still does not connect.

Towdog0352, 2022-10-04:
I am having the same issue using a Samsung Galaxy S7. I get the display on the printer to turn on but can not get octo4a to connect to the printer.

alufers, 2022-10-06:
Hi, thanks for trying octo4a!

Could you post a screenshot of the octo4a app? And a screenshot of octoprint. 

Thanks!

Gerigot, 2022-10-07:
I've an ender3v2 with the same motherboard and the only issue I had was because I was using a splitter and, while I was connecting to the printer with one and I was also charging the phone with fastcharging enabled.
what is exaclty the issue that you are having?

Towdog0352, 2022-10-07:
> Hi, thanks for trying octo4a!
> 
> Could you post a screenshot of the octo4a app? And a screenshot of octoprint. 
> 
> Thanks!

[image]

Gerigot, 2022-10-08:
I think your cable is not the right one, try to connect the one that now is connected to the printer to your phone and from the usb type A to the printer without connecting the charger ( just to see if it's working)

Towdog0352, 2022-10-08:
> I think your cable is not the right one, try to connect the one that now is connected to the printer to your phone and from the usb type A to the printer without connecting the charger ( just to see if it's working) if it connects you should see on the phone's app that the printer is connected

Thank you for the suggestion.  I swapped the cables as recommended and I was able to get it all to connect.  Do you have a recommended cable to replace the wrong one I have?

Gerigot, 2022-10-08:
Unfortunately I don't have experience with micro USB OTG cable but I think if you search for one in internet it should go auto with a micro usb male from your phone and have one USb type A female, for connecting the printer and a female USB micro for the charging

j-snijder, 2022-10-23:
Had the same problem, wrong cable :(

Buy yourself a OTG cable and watch for it that the connector that goes INTO YOUR PHONE has the OTG and not (for instance) the plug that connects to the power supply.

I bought me one from Amazon and after discovering where the problem was I bought another one and now it all works :)

wtaelewis, 2022-10-30:
j-snijder, any chance you could share the link to that OTG cable from Amazon? I've been trying to find one that works with MicroUSB phones and haven't had any luck. Thanks!

j-snijder, 2022-10-30:
Sorry but I have no idea which cable you need for micro-usb, but what I know, and told you in my response, was that you need to look for a cable that has the OTG side of the cable going into your phone, anything else will not work.

since the OTG part needs to tell the phone its there it really does have a direct connection to your phone., I guess this one would work in your case, mine was a USB-C type.

https://www.amazon.nl/Rii-F1-kabel-Micro-OTG-adapter/dp/B07SFZ33FZ/ref=sr_1_12?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2MN7JHK05VG5H&keywords=micro+usb+OTG+kabel&qid=1667153061&qu=eyJxc2MiOiIwLjAwIiwicXNhIjoiMC4wMCIsInFzcCI6IjAuMDAifQ%3D%3D&sprefix=micro+usb+otg+kabel%2Caps%2C120&sr=8-12