feelfreelinux/octo4a #295: Pixel 6 showing connection but not connecting

Digitalpeer1, 2022-10-06:
I have a pixel 6 and when I tried to connect it to my ender3V2 it is registering but not connecting pictures below

[image]
[image]
[image]

alufers, 2022-10-06:
Hi, thanks for trying octo4a!

Please add an "Additional serial port" in the Octoprint settings: ` /dev/ttyOcto4a`, and use it as the serial port. 

[image]

closed 2022-10-06 by alufers

Digitalpeer1, 2022-10-06:
Hi, how do i do that?