feelfreelinux/octo4a #309: Galaxy s7 edge

bendova-hub, 2022-10-18:
Hi I’m using a Samsung galaxy s7 edge and a y Otg cable like the one mentioned 
I can’t seem to get Otg and charging at the same time. Was wondering if anyone has a s7 and has this working and if there is anything I’m missing like a rom or something?

many help would be appreciated thanks

jakub-orbitus, 2022-10-19:
> Hi I’m using a Samsung galaxy s7 edge and a y Otg cable like the one mentioned I can’t seem to get Otg and charging at the same time. Was wondering if anyone has a s7 and has this working and if there is anything I’m missing like a rom or something?
> 
> many help would be appreciated thanks

Hi, i would suggest you mention a little more. What kind of OTG cable do you use? Do you have link where you bought it..

bendova-hub, 2022-10-19:
> Hi, i would suggest you mention a little more. What kind of OTG cable do you use? Do you have link where you bought it..

Hi yes not a problem I got it off Amazon https://a.co/d/5makiYu 
And here’s the link

embcla, 2022-10-19:
Same thing and I'm giving up: will try and keep the phone charged with a wireless pad I had lying around

bendova-hub, 2022-10-19:
> Same thing and I'm giving up: will try and keep the phone charged with a wireless pad I had lying around

I tried that kept phone alive longer but not long enough maybe I need a stronger wireless charger maybe

embcla, 2022-10-19:
Indeed you're right, I just tried and even with power-saving enabled, battery is going down (though MUCH slower than without power-saving).
I'm using a very old Samsung EP-PG920I and I suppose it's a 5W Qi1 charging pad, but can't find definitive specs.
Possibly with a 10W, the phone would stay charged

anyway, going back to the original question: I have bought 2 adapters, a simple Y and a 4-to-1 with charging switch.
The Y one just doesn't power the phone, it's meant for hard-drives probably and so it send the power to the USB-A only.
The 4-to-1 instead does send power to the phone, but the phone thinks it has a wet socket and beeps continuously until unplugged. I am not sure this is worth fixing rather than returning and finding another way (could dismantle everything and figure out why the phone is faulting, probably it gets a very low voltage and so assumes there's water to increase leakage)

bendova-hub, 2022-10-19:
> Indeed you're right, I just tried and even with power-saving enabled, battery is going down (though MUCH slower than without power-saving). I'm using a very old Samsung EP-PG920I and I suppose it's [...]

I think I’m going to get a 15+w wireless charger today and see if it’s a viable solution see if it will charge fast enough to make a difference if not return it lol

Pedrat, 2023-02-22:
Hi! Did you end up figuring something out? I'm also trying the same (although with a different phone, the Galaxy Note 3) and it seems like it's also using more power than charging unfortunately.

alufers, 2023-03-24:
Hi, Thanks for trying octo4a!
Unfortunately some phones just don't support charging and using OTG at the same time :/
This is a known problem, and unfortunately it is near-impossible to solve on many phones.

We maintain a crowd-sourced compatibility list of phones which work with octo4a:
The list: https://docs.google.com/spreadsheets/d/1pbKAELzvTrMVm6NpDpK8TfQR33w3QHQEgqkcGdbPFI0/edit?usp=sharing
Feel free to submit your phone here: https://forms.gle/UFzJebj67hCtLCBz7

Sometimes it is possible to bypass this problem by trying these solutions:

- Using wireless charging. If your phone supports it, it might be possible to use a wireless charger and use the OTG feature at the same time. Some users have resorted to gluing a wireless charger to a phone case
  [image]
  
- Hard-wiring the battery. 
  **WARNING: CONNECTING A RANDOM POWER-SUPPLY STRAIGHT TO THE BATTERY MIGHT RESULT IN A FIRE OR EXPLOSION, EXERCISE EXTREME CAUTION** 
   Some people modify their phone to charge their battery externally, bypassing the phone's battery charging circuitry altogether. Some event removed the battery (the safest option)
   [image]
- Compiling a non-standard kernel. On some models the charging-and-OTG functionality is blocked in software, and this can be bypassed by using a non-stock kernel. Please be advised that compiling an Android kernel requires some skill and a lot of time.

If you need some guidance please feel free to ask on our telegram group: (https://t.me/octo4achat)[https://t.me/octo4achat]. The community is very likely to help you.

closed 2023-03-24 by alufers