feelfreelinux/octo4a #312: Tornado Build Fail

SMASH-DIAMOND, 2022-10-31:
Hi, 
Every-time I go to build o4a it fails to build tornado. It always does it no matter what version I use.  I am on android 7 and a ASUS ZenPad.
Bullding wheels for collected packages: tornado
Building wheel for tornado (PEP 517): started
Building wheel for tornado (PEP 517): finished with status 'done'
WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted
Failed to build tornado
ERROR: Could not build wheels for tornado which use PEP 517 and cannot be installed directly

zolive33, 2022-11-01:
Hi,
I have the same issue with a Galaxy S6. No solution at this time.
Thank you for your help

Best regards.

ThePunisher, 2022-11-07:
Same on Samsung Galaxy S4 19505, Android 5.0.1 Lollipop, no SD-Card installed. Installer 1.1.2.

Scybe1, 2022-11-10:
Same problem at Samsung galaxy s5 android 6.0.1 octa4a 1.1.2 and the 3 lower ones.
I absolutely Frustes because I get no raspberry at the moment. No stock. So I thought I will made it with android. I really wish a fix for this or a guide for noobs how to fix this manually.

jbelascoainJC, 2022-11-15:
I was able to sort out this issue by Installing a custom ROM on my Galaxy A3 (2015). The stock ROM was 5.0.2 and now I'm on 7.1.2.

p2baron, 2022-11-16:
Same on LG G5. Is there anyway around this?

sashokbg, 2022-12-01:
Hello I have managed to make it work on my galaxy a3 2016 - please follow #319