feelfreelinux/octo4a #314: Installation corrupted using v1.1.2, Galaxy Tab 4 on 4.4.2

possiBri, 2022-11-23:
After running what appears to be a successful install the "continue" button at the bottom of the progress page doesn't seem to do anything. There is a notification saying the OctoPrint server is starting, and if I click on that then I get the well-known corrupted installation notice. Not sure if it's just my device, or if the previous bug is still a bug for certain devices... thanks in advance for any assistance!

arkadiusz-kozub, 2022-11-28:
Have same issue - Samsung Galaxy Tab 4

sashokbg, 2022-11-29:
Samme issue - Galaxy A3 2016 - 310f.
Tried both in stock rom as well as rooted.

Be fore the app crashes I see lots of "Operation not permitted setting attribute" messages

sashokbg, 2022-12-01:
Please check #319 I have added some information and actually managed to install and run octoprint on my galaxy a3 2016.

arkadiusz-kozub, 2022-12-04:
Had a thought maybe my tab's system is outdated. I have updated my device to Android 9, but it didn't helped :/

arkadiusz-kozub, 2023-01-01:
I have tried solution from #325 and it worked perfectly :)

alufers, 2023-03-21:
Should work in 1.2.1

closed 2023-03-21 by alufers