feelfreelinux/octo4a #316: impossible to connect

djepsylon, 2022-11-28:
hi all,
 i ve installed octo4a on my old huwawei P30pro and all works good for octoprint and connect to printer but when i try to connect to my priters who are on klipper with octoklipper, impossible to connect.
i've read the procedure to do on the help page and i've search on internet without success.
i have this error:
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x712ec29a90, open=True>(port='/dev/ttyOcto4a', baudrate=230400, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

djepsylon, 2022-11-30:
and no one for respond to us? we need help please

RickzFPV, 2022-11-30:
I think I am gonna try one of the earlier versions just to see if it is a problem with the latest version

Cass67, 2023-01-03:
Had same issue with my mate 20 pro, resolved by not using Y splitter cable, had to use a hub like this: https://amzn.to/3Ile0kA

It now connects with logs that look something to this 

before with Y splitter cable (or direct from phone to printer) 

Trying port /dev/ttyOcto4a, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

after moving to usbc hub

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyOcto4a@115200, /dev/ttyOcto4a@250000, /dev/ttyOcto4a@230400, /dev/ttyOcto4a@57600, /dev/ttyOcto4a@38400, /dev/ttyOcto4a@19200, /dev/ttyOcto4a@9600
Trying port /dev/ttyOcto4a, baudrate 115200
Connecting to port /dev/ttyOcto4a, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x71540496d0, open=True>(port='/dev/ttyOcto4a', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: µ=yok
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N1 M115*39
Recv: ok
Send: N2 M21*18
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin V1.1.4 (Jun 28 2022 14:16:31) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2 Neo EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
<snip>