feelfreelinux/octo4a #319: WARNING: Building wheel for tornado failed: [Errno 1] Operation not permitted

inv4dor, 2022-11-29:
I have a galaxy tab 3 and galaxy S5, both devices have the same error.

sashokbg, 2022-11-29:
Hello, same error on Galaxy A3 2016

I hooked the debugger and it fails on line

                    runCommand("cd Octo* && pip3 install .").waitAndPrintOutput(logger)

In class **OctoPrintHandlerRepository.kt**

sashokbg, 2022-12-01:
After tons of trials and errors I managed to get it up and running:

I added on line 144 in **OctoPrintHandlerRepository**

`runCommand("pip3 install --upgrade setuptools wheel tornado --no-cache-dir").waitAndPrintOutput(logger)`

I also changed the version of octoprint installed by octo4a to "1.7.3" by changing line 21 in **GithubRepository** to

`url("${baseUrl}repos/${repository}/releases/52578746")`

I obtained the release id by running

`curl -L https://api.github.com/repos/OctoPrint/OctoPrint/releases -H "Content-Type: application/json" | jq '.[] | select(.tag_name == "1.7.3") | .id'`

I will try installing the latest version tomorrow ...
I have no idea why wheel and tornado fail .. and why it is fixed when installing these packages manually (although there are still errors in log about wheel - it seems to fallback to installing via "setup.sh")

sashokbg, 2022-12-05:
Bad news, I broke my galaxy a3 while trying to solder some an external charging port so I will not be able to continue debugging this issue. Sorry :(

BuyMyMojo, 2022-12-10:
So I tried installing OctoPrint using the web terminal and a bunch of packages failed to install, PSUtil has an error saying it was missing gcc. could it be the lack of compilers?

chadrebuck, 2023-01-22:
> `runCommand("pip3 install --upgrade setuptools wheel tornado --no-cache-dir").waitAndPrintOutput(logger)`

pip3 is not on my moto g 7.  Is there a way to install it?

jccraig26, 2023-01-27:
My problem is similar.  I think I am in the wrong directory though.  Octoprint does not seem to start completely and I don't think SSH ever starts fully since I kept getting a connection error.  I installed simplesshd on the phone - a simple cheap tracphone - and was able to connect as root.  The problem is I am in the directory for sshd.  I am hoping that enough of Octoprint installed that the python files for pip are there if I can just get to the correct directory.  

So, can anyone point me to the directory I should be in to run that pip3 command line?

alufers, 2023-03-24:
Should be fixed in 1.2.3. Please open a new issue if the problem persists.

closed 2023-03-24 by alufers