feelfreelinux/octo4a #321: pycryptodome installation issue when installing some octoprint plugins

ramifara, 2022-12-05:
Some of the Plugins need pycryptodome to be installed and when their script tries to install pycryptodome it fails and breaks the installation. I dont have the same problem on my OctoPi installation which makes me think this is an issue with the ENV. I have also tried installing pycryptodome with SSH and the remote access extension but I keep getting the same error

I am not a Python Dev so I do not understand much from this error, hope you can help. 

`Using legacy 'setup.py install' for pycryptodome, since package 'wheel' is not installed.
Installing collected packages: pycryptodome
    Running setup.py install for pycryptodome ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-d9bvefw8/pycryptodome_94f40fae4e0140e2b342a85bf85f2ca7/setup.py'"'"'; __file__='"'"'/tmp/pip-install-d9bvefw8/pycryptodome_94f40fae4e0140e2b342a85bf85f2ca7/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-tn85ib6e/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/pycryptodome
         cwd: /tmp/pip-install-d9bvefw8/pycryptodome_94f40fae4e0140e2b342a85bf85f2ca7/
    Complete output (335 lines):
    Testing support for clang
    Target does not support clang
    Testing support for gcc
    Target does not support gcc
    Testing support for stdint.h header
    Target does not support stdint.h header
    Testing support for 128-bit integer
    Target does not support 128-bit integer
    Testing support for cpuid.h header
    Target does not support cpuid.h header
    Testing support for intrin.h header
    Target does not support intrin.h header
    Testing support for posix_memalign
    Target does not support posix_memalign
    Testing support for memalign
    Target does not support memalign
    Testing support for SSE2(intrin.h)
    Target does not support SSE2(intrin.h)
    Testing support for SSE2(x86intrin.h)
    Target does not support SSE2(x86intrin.h)
    Testing support for SSE2(emmintrin.h)
    Target does not support SSE2(emmintrin.h)
    Warning: compiler does not support AESNI instructions
    Warning: compiler does not support CLMUL instructions
    running install
    running build
    running build_py
    creating build/lib.linux-aarch64-3.9
    creating build/lib.linux-aarch64-3.9/Crypto
    copying lib/Crypto/__init__.py -> build/lib.linux-aarch64-3.9/Crypto
    creating build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/Salsa20.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/DES.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_EKSBlowfish.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/PKCS1_OAEP.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_siv.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_ecb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_openpgp.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_ofb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/ARC2.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/AES.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_ocb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_gcm.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_cbc.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/Blowfish.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/ChaCha20.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/ARC4.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/ChaCha20_Poly1305.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/DES3.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/PKCS1_v1_5.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_ccm.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_ctr.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_cfb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/CAST.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/_mode_eax.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    copying lib/Crypto/Cipher/__init__.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
    creating build/lib.linux-aarch64-3.9/Crypto/Hash
    copying lib/Crypto/Hash/BLAKE2s.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
    copying lib/Crypto/Hash/Poly1305.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
[... 253 lines cut ...]
    copying lib/Crypto/Util/Padding.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/_raw_api.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/strxor.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/Counter.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/RFC1751.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/number.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/asn1.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/_cpu_features.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Util/_file_system.pyi -> build/lib.linux-aarch64-3.9/Crypto/Util
    copying lib/Crypto/Math/_IntegerBase.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    copying lib/Crypto/Math/Numbers.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    copying lib/Crypto/Math/_IntegerGMP.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    copying lib/Crypto/Math/_IntegerNative.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    copying lib/Crypto/Math/Primality.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    copying lib/Crypto/Math/_IntegerCustom.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
    running build_ext
    building 'Crypto.Hash._MD2' extension
    creating build/temp.linux-aarch64-3.9
    creating build/temp.linux-aarch64-3.9/src
    gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/usr/include/python3.9 -c src/MD2.c -o build/temp.linux-aarch64-3.9/src/MD2.o
    gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
    compilation terminated.
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit `

alufers, 2023-03-24:
You need to install Plugin extras from the app's settings.

closed 2023-03-24 by alufers

sv0o, 2023-06-11:
@alufers 
despite having plugin extras installed (from way earlier), I seem to be having this issue while attempting to install the OctoApp companion plugin.  The plugin had been working before I updated Octoprint to 1.9.0  and this new Octoprint version had some major updates that removed support for older stuff.  I'm attaching log files in hopes that something obvious will be apparent. 

2023-06-09 15:22:25,543 > Building wheels for collected packages: OctoApp, pycryptodome
2023-06-09 15:22:25,544 > Building wheel for OctoApp (setup.py): started
2023-06-09 15:22:39,829 > Building wheel for OctoApp (setup.py): finished with status 'done'
2023-06-09 15:22:39,830 > Created wheel for OctoApp: filename=OctoApp-1.2.3-py2.py3-none-any.whl size=27679 sha256=a12620451584adcc4f39a20cf3acacd54451ef13bd089fa8c75d3177c3196667
2023-06-09 15:22:39,831 > Stored in directory: /tmp/pip-ephem-wheel-cache-ltuad53b/wheels/3a/6c/2f/ba75507d5c5dfd9b3e3b5a7ce5eaa810f0276fdd08347ca814
2023-06-09 15:22:39,832 > Building wheel for pycryptodome (setup.py): started
2023-06-09 15:22:58,279 > Building wheel for pycryptodome (setup.py): finished with status 'error'
2023-06-09 15:22:58,279 > Running setup.py clean for pycryptodome
2023-06-09 15:22:58,809 ! error: subprocess-exited-with-error
2023-06-09 15:22:58,809 ! 
2023-06-09 15:22:58,810 ! × python setup.py bdist_wheel did not run successfully.
2023-06-09 15:22:58,810 ! │ exit code: 1
2023-06-09 15:22:58,811 ! ╰─> [335 lines of output]
2023-06-09 15:22:58,811 ! Testing support for clang
2023-06-09 15:22:58,812 ! Target does not support clang
2023-06-09 15:22:58,812 ! Testing support for gcc
2023-06-09 15:22:58,813 ! Target does not support gcc
2023-06-09 15:22:58,813 ! Testing support for stdint.h header
2023-06-09 15:22:58,814 ! Target does not support stdint.h header
2023-06-09 15:22:58,814 ! Testing support for 128-bit integer
2023-06-09 15:22:58,815 ! Target does not support 128-bit integer
2023-06-09 15:22:58,815 ! Testing support for cpuid.h header
2023-06-09 15:22:58,816 ! Target does not support cpuid.h header
2023-06-09 15:22:58,816 ! Testing support for intrin.h header
2023-06-09 15:22:58,816 ! Target does not support intrin.h header
2023-06-09 15:22:58,817 ! Testing support for posix_memalign
2023-06-09 15:22:58,817 ! Target does not support posix_memalign
2023-06-09 15:22:58,818 ! Testing support for memalign
2023-06-09 15:22:58,818 ! Target does not support memalign
2023-06-09 15:22:58,818 ! Testing support for SSE2(intrin.h)
2023-06-09 15:22:58,819 ! Target does not support SSE2(intrin.h)
2023-06-09 15:22:58,819 ! Testing support for SSE2(x86intrin.h)
2023-06-09 15:22:58,819 ! Target does not support SSE2(x86intrin.h)
2023-06-09 15:22:58,820 ! Testing support for SSE2(emmintrin.h)
2023-06-09 15:22:58,820 ! Target does not support SSE2(emmintrin.h)
2023-06-09 15:22:58,820 ! Warning: compiler does not support AESNI instructions
2023-06-09 15:22:58,821 ! Warning: compiler does not support CLMUL instructions
2023-06-09 15:22:58,821 ! running bdist_wheel
2023-06-09 15:22:58,821 ! running build
2023-06-09 15:22:58,822 ! running build_py
2023-06-09 15:22:58,823 ! creating build/lib.linux-armv7l-cpython-310
2023-06-09 15:22:58,824 ! creating build/lib.linux-armv7l-cpython-310/Crypto
2023-06-09 15:22:58,824 ! copying lib/Crypto/__init__.py -> build/lib.linux-armv7l-cpython-310/Crypto
2023-06-09 15:22:58,825 ! creating build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,825 ! copying lib/Crypto/Cipher/_mode_ctr.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,826 ! copying lib/Crypto/Cipher/_mode_openpgp.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,826 ! copying lib/Crypto/Cipher/PKCS1_v1_5.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,827 ! copying lib/Crypto/Cipher/_mode_gcm.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,827 ! copying lib/Crypto/Cipher/_mode_ofb.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,827 ! copying lib/Crypto/Cipher/DES3.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,828 ! copying lib/Crypto/Cipher/Salsa20.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,828 ! copying lib/Crypto/Cipher/__init__.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,828 ! copying lib/Crypto/Cipher/Blowfish.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,828 ! copying lib/Crypto/Cipher/ARC4.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,829 ! copying lib/Crypto/Cipher/_mode_eax.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,829 ! copying lib/Crypto/Cipher/PKCS1_OAEP.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,829 ! copying lib/Crypto/Cipher/_mode_ocb.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,829 ! copying lib/Crypto/Cipher/DES.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,829 ! copying lib/Crypto/Cipher/ChaCha20.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,830 ! copying lib/Crypto/Cipher/_mode_cbc.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
2023-06-09 15:22:58,830 ! copying lib/Crypto/Cipher/_mode_cfb.py -> build/lib.linux-armv7l-cpython-310/Crypto/Cipher
[... cut ...]
2023-06-09 15:22:58,903 ! gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
2023-06-09 15:22:58,903 ! compilation terminated.
2023-06-09 15:22:58,903 ! error: command 'gcc' failed with exit code 1
2023-06-09 15:22:58,904 ! [end of output]
2023-06-09 15:22:58,904 ! 
2023-06-09 15:22:58,904 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2023-06-09 15:22:58,904 ! ERROR: Failed building wheel for pycryptodome
2023-06-09 15:23:09,863 > Successfully built OctoApp
2023-06-09 15:23:09,864 > Failed to build pycryptodome

sv0o, 2023-06-11:
here's an apology and an update.  Sorry, I just realized 1.2.5 octo4a was available and I installed that apk.  

same errors

**_then I re-read issue #325 and tried this in ssh:_**

pip3 install --upgrade setuptools wheel tornado --no-cache-dir
Requirement already satisfied: setuptools in /usr/lib/python3.10/site-packages (65.6.0)
Collecting setuptools
  Downloading setuptools-67.8.0-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 838.6 kB/s eta 0:00:00
Requirement already satisfied: wheel in /usr/lib/python3.10/site-packages (0.40.0)
Requirement already satisfied: tornado in /usr/lib/python3.10/site-packages (6.2)
Collecting tornado
  Downloading tornado-6.3.2.tar.gz (508 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 508.8/508.8 kB 1.1 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: tornado
  Building wheel for tornado (pyproject.toml) ... done
  Created wheel for tornado: filename=tornado-6.3.2-cp38-abi3-linux_armv7l.whl size=420634 sha256=40acc020bee0dd1871961125e2345035cf550db5e61906d693120ab72ebd2f2d
  Stored in directory: /tmp/pip-ephem-wheel-cache-2rmojua2/wheels/f3/2d/56/d03495a168e3191397682d2b832fa5733a7acacc86454ec9a2
Successfully built tornado
Installing collected packages: tornado, setuptools
  Attempting uninstall: tornado
    Found existing installation: tornado 6.2
    Uninstalling tornado-6.2:
      Successfully uninstalled tornado-6.2
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.6.0
    Uninstalling setuptools-65.6.0:
      Successfully uninstalled setuptools-65.6.0

**_then, I attempted this:_**

pip3 install pycryptodome
Collecting pycryptodome
  Downloading pycryptodome-3.18.0.tar.gz (4.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.6/4.6 MB 4.9 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: pycryptodome
  Building wheel for pycryptodome (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [335 lines of output]
      Testing support for clang
      Target does not support clang
      Testing support for gcc
      Target does not support gcc
      Testing support for stdint.h header
      Target does not support stdint.h header
      Testing support for 128-bit integer
      Target does not support 128-bit integer
      Testing support for cpuid.h header
      Target does not support cpuid.h header
      Testing support for intrin.h header
      Target does not support intrin.h header
      Testing support for posix_memalign
      Target does not support posix_memalign
      Testing support for memalign
      Target does not support memalign
      Testing support for SSE2(intrin.h)
      Target does not support SSE2(intrin.h)
      Testing support for SSE2(x86intrin.h)
      Target does not support SSE2(x86intrin.h)
      Testing support for SSE2(emmintrin.h)
      Target does not support SSE2(emmintrin.h)
[... 661 lines cut ...]
      copying lib/Crypto/Math/Primality.pyi -> build/lib.linux-armv7l-cpython-310/Crypto/Math
      copying lib/Crypto/Math/_IntegerGMP.pyi -> build/lib.linux-armv7l-cpython-310/Crypto/Math
      copying lib/Crypto/Math/Numbers.pyi -> build/lib.linux-armv7l-cpython-310/Crypto/Math
      copying lib/Crypto/Math/_IntegerNative.pyi -> build/lib.linux-armv7l-cpython-310/Crypto/Math
      copying lib/Crypto/Math/_IntegerBase.pyi -> build/lib.linux-armv7l-cpython-310/Crypto/Math
      running build_ext
      building 'Crypto.Hash._MD2' extension
      creating build/temp.linux-armv7l-cpython-310
      creating build/temp.linux-armv7l-cpython-310/src
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -Os -fomit-frame-pointer -g -O2 -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=32 -DLTC_NO_ASM -Isrc/ -I/usr/include/python3.10 -c src/MD2.c -o build/temp.linux-armv7l-cpython-310/src/MD2.o
      gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
      compilation terminated.
      error: command 'gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pycryptodome

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

**_So, apart from having the latest tornado and wheel and octo4a, I seem to be in the same status as my last post._**