feelfreelinux/octo4a #322: can not install the files in the app

Ferre-Defossez, 2022-12-06:
I have a Redmi 9a where I want to install the app. the .apk I get installed.
i on the app it says no existing installs detected. so i install octoprint and it gives the message "missing storage permisison" i get a popup for the app to give access to the to the photos, media and files on the cell phone. next it goes to the screen for installation. on step 1(instaling bootrap) and sometimes on step 2(instaling octoprint) the app crashes. If I open the app again then I go to the screen of octoprint where I see the status of the files , printer and camera. here it says that the installation is corrupted "installation corrupted tap to reinstall the app" with a broken heart next to it. if I push the broken heart the app crashes again and starts all over again.
i used the last version 1.1.2

maybe i am doing something wrong? maybe the app is too heavy for the cell phone? can anyone give suggestions what i can do.

albertsj1, 2022-12-08:
The same thing happens for me, except I'm trying to run this on a Fire Tablet HD 10 Plus, which I realize may not be supported at all.  I just thought I'd give it a try.

RickzFPV, 2022-12-09:
Yeah I am having the exact same problem with a BLU cellphone that was recommended specifically for this I was able to get the latest version up and running on my Lenovo Tablet, but now gives issues with this phone

alufers, 2023-03-21:
Please try the 1.2.1 version.

closed 2023-03-21 by alufers