feelfreelinux/octo4a #325: Install crashes and corrupts every time at the same point "python setup.py egg_info

stayingarmed, 2022-12-09:
A few months ago this program worked fine on my Galaxy Note 8 but then there was an update on my phone that restarted it and at this point it never installs and always corrupts.

[image]

Jano952, 2022-12-11:
I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix in #319. Here is how to do it using SSH commands, so you dont have to modify the .apk!

1. Install Octo4A, let it install OctoPrint. Let the setup go, let it crash.
2. Start the Octo4A, you can see it is corrupted, but dont reinstall !
3. Go to Config, enable OpenSSH and set Port and Password.
4. Log into SSH from your computer or whatever. (ssh root@IP -p PORT)
5. Now enter command, always wait for finish: 
_pip3 install --upgrade setuptools wheel tornado --no-cache-dir_
6. I am not sure with the next command, it is probably not needed, but i put it in:
_curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s_
7. Then enter this, once asked, overwrite all files:
_unzip octoprint.zip_
8. And finally this command (ALSO COPY THE DOT):
_cd Octo* && pip3 install ._
9. I also installed plugin extras from the setting menu
10. Now Force stop the app somehow (I needed to go to app setting and force stop on my Huaweii P20 Pro, I did similar on Redmi Note 4)
11. Turn on the Octo4A, and it should work !

TheDooblem, 2022-12-11:
Thank You! This worked. If only I had found this before factory resetting my phone and forgetting to save the backup. Oh well.

epsilontheta, 2022-12-11:
@Jano952 can confirm that this is working for me also. Thank you!

KubaArrow, 2022-12-11:
@Jano952 You are amazing :smile: 
On Redmi Note 5 working

drhealx, 2022-12-11:
Im having the same Problem on a Redmi Note 5 and cant get past this step. Where do i see the needed IP Adress from the phone?

Log into SSH from your computer or whatever. (ssh root@IP -p PORT)

KubaArrow, 2022-12-11:
@drhealx In details wifi connect or DHCP clients in router panel

Jano952, 2022-12-11:
@drhealx did you force stop Octo4A? It checks for succesfull instalation only after you totally stop the app. I had to go to Android settings -> apps -> Octoprint for Android and force stop it, otherwise it kept running in background somehow.

drhealx, 2022-12-11:
@Jano952 yes, i stopped it by tapping on "App info" -> "force stop". But i recognized that in the last step 8. "And finally this command (ALSO COPY THE DOT):
cd Octo* && pip3 install ." i got an error message that said something like:
"dont know the command 'install.' did you mean 'install'" and i just ignored that message, maybe thats the problem

stayingarmed, 2022-12-11:
Can confirm....this worked for me as well! Thank you for coming up with this solution!!!!! Now I can use my Galaxy Note 8 to run my printer instead of this old raspberry pi!

closed 2022-12-11 by stayingarmed

BertrandTheHealer, 2022-12-11:
I wasn't getting an error, the app was just crashing partway through installation and when I reopened it I would get the "Installation Corrupted" text.

[Jano952](https://github.com/Jano952)'s instructions fixed it for me.

micah-simmler, 2022-12-11:
Followed and seems to be working for me. Thanks!

RickzFPV, 2022-12-11:
@Jano952 hey is there any chance you could give better directions on how to do this for someone that pretty much knows nothing about android but can follow directions really well

Jano952, 2022-12-11:
@RickzFPV Once you setup the SSH it is easy job (In Win10 you can simply use command prompt to run ssh). Sadly I dont have blog, but maybe I can create video about it. I hope the install script will simply get fixed in coming days and nothing will have to be changed.

micah-simmler, 2022-12-11:
> @Jano952 hey is there any chance you could give better directions on how to do this for someone that pretty much knows nothing about android but can follow directions really well. The thing I cant understand or figure out is what you mean is #4 of your instructions. I can ssh into my raspberry pi's with no issue but do not understand at all on android

Where are you stuck, enabling SSH? It is an option in settings will prompt for a password. Install Putty and SSH to your Phone IP with port 8022 which was the default.

Once you get connected username is root with the password you specified then give it these commands one at a time.

pip3 install --upgrade setuptools wheel tornado --no-cache-dir
curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s
unzip octoprint.zip
cd Octo* && pip3 install .

Install Plugin Extras option on the phone

Go into App Settings and Force Close the App then you should be good.

RickzFPV, 2022-12-11:
@micah-simmler Thank you So much Got it going on the tablet.  I have one of the blue Tracphones running android 11that I would also like to set it up on.  when I try and log in through putty it tells me refused.  is there a setting I need to change in the phone to get it going

micah-simmler, 2022-12-11:
> @micah-simmler Thank you So much Got it going on the tablet. I have one of the blue Tracphones running android 11that I would also like to set it up on. when I try and log in through putty it tells me refused. is there a setting I need to change in the phone to get it going

Just make sure It is enabled nd you are on the right port should be it not sure what else it could be. I just tried it on mine without SSH Enabled and it says connection refused. I know Putty Defaults to port 22 for SSH but Octo4a is 8022

RickzFPV, 2022-12-11:
ok I will give it a try again

RickzFPV, 2022-12-11:
@micah-simmler is there something I am missing with a setting on the phone or something with this Blue view 2 it keeps kicking back a network error connection refused.  The tablet is good to go but can't for the life of me get into the phone

Saurabh251-cmd, 2022-12-12:
worked here too thx

thomasbra1, 2022-12-12:
> @micah-simmler is there something I am missing with a setting on the phone or something with this Blue view 2 it keeps kicking back a network error connection refused. The tablet is good to go but can't for the life of me get into the phone

I tried to do this on a B131DL BV2 and after the failed install it will try to enable ssh but then immediately kill it. So it seems it can't install enough to try the fix on it.

SR10638, 2022-12-13:
If your getting the SSH network error-Connection refused via whatever client (PuTTY/Terminal/CMD),you can try via your browser directly. I found going to the address http://(PHONES IP ADDRESS):5002/ will prompt for the ssh credentials you set in the config on the app (user:root, pass:whatever you set). You'll be viewing a live terminal session from the linux OS the app sets up. You can enter the commands directly on this page just like in an SSH client. Just continue the above instructions from step 5 down and you should be good to go.  Note: If your going to copy paste the commands, when pasting into the web terminal use: Ctrl+Shift+V. Also remember that the Octoprint's web console for management is on port 5000 NOT 5002, so managing Octoprint will be at http://(PHONES IP ADDRESS):5000/.

Lilbrulesall, 2022-12-13:
> I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do it using SSH commands, so you dont have to modify the .apk!
> 
> 1. Install Octo4A, let it install OctoPrint. Let the setup go, let it crash.
> 2. Start the Octo4A, you can see it is corrupted, but dont reinstall !
> 3. Go to Config, enable OpenSSH and set Port and Password.
> 4. Log into SSH from your computer or whatever. (ssh root@IP -p PORT)
> 5. Now enter command, always wait for finish:
>    _pip3 install --upgrade setuptools wheel tornado --no-cache-dir_
> 6. I am not sure with the next command, it is probably not needed, but i put it in:
>    _curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s_
> 7. Then enter this, once asked, overwrite all files:
>    _unzip octoprint.zip_
> 8. And finally this command (ALSO COPY THE DOT):
>    _cd Octo* && pip3 install ._
> 9. I also installed plugin extras from the setting menu
> 10. Now Force stop the app somehow (I needed to go to app setting and force stop on my Huaweii P20 Pro, I did similar on Redmi Note 4)
> 11. Turn on the Octo4A, and it should work !

For some reason, when I get to step 7, it says operation not permitted. Does anyone know why this would be? I get tthe prompt to replace and tell it "A" to replace them all, but each one fails saying  "(warning) cannot set modif./access times" "Operation Not Permitted"

ZevaKikel, 2022-12-16:
> I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do it using SSH commands, so you dont have to modify the .apk!
> ....

I'm not going to ask you how you managed to get this fix, but you have all my admiration! It has worked for me too!!! Thank you!!!

kerad-fly, 2022-12-17:
@Jano952 thx for advice.

Jano952, 2022-12-17:
> I'm not going to ask you how you managed to get this fix, but you have all my admiration! It has worked for me too!!! Thank you!!!

It was a coincidence that two days earlier I was having very hard time installing Python through SSH into my very old NAS. So I had installation of these damn Python packages fresh in memory 😆 
I saw @sashokbg fix in #319 and I was like...  huh, those are just normal commands for shell... and Octo4A has SSH, maybe I can send them that way instead of recompiling the code ?

BCorbett0312, 2022-12-17:
I can SSH in without any issue but when i try to run step 5 it goes through every version of Tornado and still errors out saying no dependency fits the requirements

calvissuperman, 2022-12-17:
Wasn't able to login for a LONG time until I forwarded my phones ssh server port. Hope this helps anyone else who's having troubles

sanjiv204, 2022-12-19:
> If your getting the SSH network error-Connection refused via whatever client (PuTTY/Terminal/CMD),you can try via your browser directly. I found going to the address http://(PHONES IP ADDRESS):5002/ [...]

This is the best and easiest approach for an uninitiated member like me. Thanks

BCorbett0312, 2022-12-19:
I was finally able to get it installed.  I used those steps on 325 but I had to use the web browser on my desktop to SSH in.  I tried putty and termius but it wouldnt let me log in as root which prevented the script from running.  After getting the web browser shell up and logging in as root I was able to execute the scripts.  I tried this on 3 different samsung phones (S8, S9, and S9+) and it worked for each one. 

The other terminal apps just wouldnt let me log in as root.

luigime5, 2022-12-20:
[image]

I tried @Jano952s commands and I got this error after doing step 6. I have an LG g6 and was using putty.

BCorbett0312, 2022-12-20:
> [image]
> 
> I tried @Jano952s commands and I got this error after doing step 6. I have an LG g6 and was using putty.

This is the issue I was having too. I ended up enabling the remote terminal and then it would let me log in as root and by pass all the permission errors

luigime5, 2022-12-20:
Thank you @BCorbett0312 that fixed it.

Escrich, 2022-12-21:
> > I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do it using SSH commands, so you dont have to modify the .apk!
> > 
> > 1. Install Octo4A, let it install OctoPrint. Let the setup go, let it crash.
> > 2. Start the Octo4A, you can see it is corrupted, but dont reinstall !
> > 3. Go to Config, enable OpenSSH and set Port and Password.
> > 4. Log into SSH from your computer or whatever. (ssh root@IP -p PORT)
> > 5. Now enter command, always wait for finish:
> >    _pip3 install --upgrade setuptools wheel tornado --no-cache-dir_
> > 6. I am not sure with the next command, it is probably not needed, but i put it in:
> >    _curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s_
> > 7. Then enter this, once asked, overwrite all files:
> >    _unzip octoprint.zip_
> > 8. And finally this command (ALSO COPY THE DOT):
> >    _cd Octo* && pip3 install ._
> > 9. I also installed plugin extras from the setting menu
> > 10. Now Force stop the app somehow (I needed to go to app setting and force stop on my Huaweii P20 Pro, I did similar on Redmi Note 4)
> > 11. Turn on the Octo4A, and it should work !
> 
> For some reason, when I get to step 7, it says operation not permitted. Does anyone know why this would be? I get tthe prompt to replace and tell it "A" to replace them all, but each one fails saying  "(warning) cannot set modif./access times" "Operation Not Permitted"
> 
> EDIT: I continued to follow the steps and ignored the errors and it works great now. Thank you so much for this!
> 

This really work

RickzFPV, 2022-12-21:
It does work but only on certain phones and tablets with certain versions of Android.  I can verify it worked fine on my 2 Lenovo tablets running Android 10.  I have not been able to get it working on 2 different phones with Android 11

TheHunter3398, 2022-12-26:
Can some one explain this in idiot terms to me I have no experience with this kind of thing and have been struggling. Thanks

TheHunter3398, 2022-12-27:
Never mind I figured it out

Midgar357, 2022-12-27:
> I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do [...]

This worked awesome thanks!

aravindkarthik96, 2022-12-27:
> pip3 install --upgrade setuptools wheel tornado --no-cache-dir

this command returns the following error
`-ash: pip3: not found`

anything I can do to fix this?

thebearjew523, 2022-12-28:
> One thing to look for is make sure you know what IP your phone was assigned by default they randomize the MAC address and I kept getting different IP addresses until I set it to use my device MAC.

What do you mean by this? I'm still unable to ssh into the server even when verifying my IP on my router page. I still get "Connection Refused". Thanks in advance!

Sejkorka, 2022-12-28:
> 

Hi when i try to install i get this:
-bash: pip3: command not found

RickzFPV, 2022-12-28:
What version of android is the device on and what device are you trying to install it on

Sejkorka, 2022-12-28:
> What version of android is the device on and what device are you trying to install it on

Lenovo k5 plus on Android 10 
Version 0.0.5 is working

RickzFPV, 2022-12-28:
It should work then I had it running on 2 of my Lenovo tablets running android 10 I have recently moved away from all octoprint installs completely because of all the issues recently I have 2 now running on mainsail and klipper and will be adding the rest to a Linux server running the same

RickzFPV, 2022-12-28:
@Sejkorka I honestly don't know why you can't get it up and running, for me I have backed away from this program until there is a permanent stable fix for it.  I have invested too much money in equipment and time in this program, to make it worthwhile anymore, If a new working version comes out I will maybe look at it again in the future.

dzidza, 2022-12-29:
> I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do [...]

This worked for me as well. 
It showed a lot of errors during those SSH commands, so I was skeptical, but it worked after all :)

Ed2275, 2022-12-29:
Hello everyone, hoping to get some help with the SSH connection.

I tried Putty, CMD on Win11 and the browser option. I'm always getting the "Connection Refused" message.

I'm at a loss and not sure what to do next.

sashokbg, 2022-12-29:
@Jano952 can you please open a Pull Request with your changes so that people who cannot SSH into their phones (@Ed2275 @RickzFPV) get the fix as well ?

If everything goes fine the project's maintainer will hopefully merge it @feelfreelinux  

@Ed2275 @RickzFPV
Worse case scenario I can package an APK for you guys but I guarantee absolutely nothing and you will be installing an APK from a complete stranger on the internet on your phones :laughing:

RickzFPV, 2022-12-29:
@Ed2275 i will definitely test an APL for you I don’t have anything to lose lol.  I am really good with most OS’s except Android so I would be more than willing to test an APL for you.  Just so you know equipment will be BLU B131DL phone running Android 11

Ed2275, 2022-12-29:
> @Ed2275 i will definitely test an APL for you I don’t have anything to lose lol. I am really good with most OS’s except Android so I would be more than willing to test an APL for you. Just so you know equipment will be BLU B131DL phone running Android 11

@RickzFPV That would be awesome. Appreciate your effort. I'm trying to get it running on a NUU A9L, it came with Android 12. Not sure I can down grade? First android phone I have ever used and got it just to run Octoprint 😆

RickzFPV, 2022-12-29:
@Ed2275 that is the exact reason I bought the blu phones lol specially for this

Darkwhisp3r, 2022-12-30:
[image]
Has anyone seen this problem before?

TheNormWorks, 2022-12-30:
> I am not gonna lie, I am noob, but somehow I made it working. I did steps I saw in _OctoPrintHandlerRepository.kt_ file between lines 122-144 and used the fix by @sashokbg in #319. Here is how to do [...]

Thanks so much Jano925! This worked for me.
Some notes on my experience...
I'm using an LG G Pad III 8.0 I picked up for $40. I was interested in this tablet specifically because it has two USB ports....a mini port + a standard A port. It has an 8-core processor, 2G RAM and running Android 7.0. It's serving a Creality CR-6 SE running the CR-6 SE Community Edition software.
I followed the instructions above but skipped step #6 and Octo4a worked immediately once completed.
I'm using the USB-mini port to provide power to the tablet and the USB-A port for connection to the printer - the USB-A port also provides power out so I've taped over one of the power pins because the CR-6 SE board does not like power injected into it's USB port (who does...am i right?). Also, the LG G Pad III provides data connection only on the first connected device so I made sure to connect the printer cable first and then the power cable. OctoPrint Serial Port is set to /dev/ttyOcto4a and baud is set to 115200. I hope this helps anyone looking to use a similar setup.
My first print using Octo4a was a 10-hour phone stand which just completed. It's a perfect print with no artifacts of any kind. Now to replace my old Mac Mini which is precariously serving my other printer....I'm afraid that hodgepodge setup will break on a future OctoPrint update.

FKentD, 2022-12-30:
Thanks dearly, this worked for me and my Samsung S7

futhgar, 2022-12-31:
Trying to install this on a Galaxy Note 10+. I am unable to SSH into the phone via browser, PuTTY or command/powershell Win 10. I keep getting a Permission Denied even after trying several different passwords. Any help would be appreciated.

FKentD, 2022-12-31:
Using Terminal on Mac I was able to SSH straight into my S7. I did disable all Samsung security settings and anything else I could find that had anything to do with security. Not sure if that’s what enabled me to use SSH or not, it’s been a long while since I’ve used android period and haven’t found the need to use these tools on the only windows machine I use at work. Besides, their systems wouldn’t allow it anyway. Good luck. Sent from my iPhone

mordenm1, 2022-12-31:
Like a dumb I'm sitting there hitting y enter y enter not noticing A was there....

MissingThumb, 2022-12-31:
got everything working up to step 5, it just goes through versions of tornado and then decides that none are suitable for download. I'm using a Oneplus 8t, cause i just got a pixel 7 and figured id just use my old phone. worth mentioning that it's on Android 12, so I got no idea if its even possible.

Melbatoast1985, 2023-01-01:
What’s an ssh? And how do I use it. I’m an extra double probation noob.

mordenm1, 2023-01-01:
> What’s an ssh? And how do I use it. I’m an extra double probation noob.

Oh boy buddy. So check into your settings on the octo4a app. Turn on open ssh. Note the port number. Go to your wifi settings(on the phone your using for octo4a) note your ip address. You can youtube how to ssh into a linux machine. This is how I do it. In windows open start menu find and open powershell. Type in `ssh root@<yourip> -p <your port>` It will ask for the password. Type in the password you set when turning on ssh in the app. If this isn't helping enough definitely youtube some ssh help guides.

Melbatoast1985, 2023-01-01:
You’re a gem thanks I’ll try this in the morning.

[84 more replies]