feelfreelinux/octo4a #326: Installation corrupted

zdog90210, 2022-12-10:
I have tried as many different fixes I could think of including 3 different phones and one tablet as well as upgrading and downgrading my pixel 2 xl to see if various android versions were the culprit no luck every time I install it gives me the installation corrupted. I have tried all versions and the only working one is the 0.0.5 release none of the main releases work. I have used the app flawlessly for months prior to updating my phone and accidentally formatted my user data. I have tried on both rooted and unrooted phones and no luck. is there any way to chose the install version of octoprint to try if that is the error. I am not sure what logs I can attach as the app crashes after installation.

sybershot, 2022-12-10:
Not feeling like creating yet another issue regarding corrupted installations, having same issue there, the only working version for me is 0.0.5 running on rooted samsung galaxy a02. I've managed to extract some logs from 1.1.2 though. 
<details>
  <summary>Log from 1.1.2</summary>
  No bootstrap detected, proceeding with installation
 Downloading bootstrap 0.1.8
 Bootstrap extracted, setting it up...
 add-user.sh
 fake_proc_stat
 install-bootstrap.sh
 ioctlHook.so
 root
 root-pre5
 rootfs.tar.xz
 run-bootstrap.sh
 bootstrap ready, run with run-bootstrap.sh
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 Welcome to Alpine!
 
 The Alpine Wiki contains a large amount of how-to guides and general
 information about administrating Alpine systems.
 See <http://wiki.alpinelinux.org/>.
 
 You can setup the system with the command: setup-alpine
 
 You may change this message by editing /etc/motd.
 
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 /bin/sh: apk: not found
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 /bin/sh: can't create /etc/ssh/sshd_config: nonexistent directory
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 /bin/sh: ssh-keygen: not found
 Bootstrap installation done
 Bootstrap installed
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 /bin/sh: curl: not found
 /bin/sh: bash: not found
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 /bin/sh: curl: not found
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
 unzip: can't open octoprint.zip[.zip]
 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory

</details>

stayingarmed, 2022-12-10:
Having the same issue on my galaxy note 8. I’ve tried rooting it, installing about 8 different roms covering 5 different Android versions, every version of the app and only 0.0.5 works for me as well. It crashes and when I open it back up it’s at the “disable battery optimization” pop up and once I click that to disable optimized battery and click to restart the install again, it just crashes again. Rinse and repeat.

KevinDW007, 2022-12-10:
Same for me on a Samsung Galaxy S6. Device is not rooted, but that isnt a requirement afaik?

Version 0.0.5 also seems to work (havent tested, but I get the octoprint is running)

TimeCRC, 2022-12-10:
Same here with my S10 running Android 10.

epsilontheta, 2022-12-10:
Same problem here with Motorola G5 plus running Android 11.
Seems like the app is crashing during installation.

micah-simmler, 2022-12-11:
Same issue on a Pixel 3 XL was working then had to reinstall and now will not work also on a Galaxy A12 and PIxel 5

Saurabh251-cmd, 2022-12-11:
same issue on Moto e3 power android 7.1 rooted

hyzx86, 2022-12-11:
same issue on 1.1.1 when  installing dependencies , the app will shutdown

hyzx86, 2022-12-11:
I tried 1.1.1, 1.1.2
And 1.0.2 failed
Finally, the log was saved when the 1.0.2 execution failed

``` LOG
  inflating: OctoPrint-OctoPrint-ab3ebf1/versioneer.py  
🐸 proot warning: can't sanitize binding "/data/data/com.octo4a/files/serialpipe": No such file or directory
🐸 Processing /root/OctoPrint-OctoPrint-ab3ebf1
🐸     ERROR: Command errored out with exit status 1:
🐸      command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-6v5wr_mn/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-6v5wr_mn/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-yca5tos6
🐸          cwd: /tmp/pip-req-build-6v5wr_mn/
🐸     Complete output (31 lines):
🐸     Traceback (most recent call last):
🐸       File "<string>", line 1, in <module>
🐸       File "/usr/lib/python3.9/site-packages/setuptools/__init__.py", line 16, in <module>
🐸         import setuptools.version
🐸       File "/usr/lib/python3.9/site-packages/setuptools/version.py", line 1, in <module>
🐸         import pkg_resources
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3243, in <module>
🐸         def _initialize_master_working_set():
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3226, in _call_aside
🐸         f(*args, **kwargs)
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
🐸         working_set = WorkingSet._build_master()
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 559, in _build_master
🐸         ws = cls()
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 552, in __init__
🐸         self.add_entry(entry)
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 608, in add_entry
🐸         for dist in find_distributions(entry, True):
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2059, in find_on_path
🐸         path_item_entries = _by_version_descending(filtered)
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2029, in _by_version_descending
🐸         return sorted(names, key=_by_version, reverse=True)
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in _by_version
🐸         return [packaging.version.parse(part) for part in parts]
🐸       File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in <listcomp>
🐸         return [packaging.version.parse(part) for part in parts]
🐸       File "/usr/lib/python3.9/site-packages/packaging/version.py", line 52, in parse
🐸         return Version(version)
🐸       File "/usr/lib/python3.9/site-packages/packaging/version.py", line 197, in __init__
🐸         raise InvalidVersion(f"Invalid version: '{version}'")
🐸     packaging.version.InvalidVersion: Invalid version: '/usr/lib/python3.9/site'
🐸     ----------------------------------------
🐸 WARNING: Discarding file:///root/OctoPrint-OctoPrint-ab3ebf1. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
🐸 ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
💻 Dependencies installed
```

[1.0.2.log](https://github.com/feelfreelinux/octo4a/files/10202214/1.0.2.log)

TimeCRC, 2022-12-11:
Hey guys, try the fix from here https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359 .
Fixed it for me!

BertrandTheHealer, 2022-12-11:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

Can confirm this worked for me as well.

sybershot, 2022-12-11:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

Sadly, this didn't work for me, i assume octo4a doesn't setup alpine container where i could ssh into. The only other option for me is to modify the sources or use 0.0.5.

Jano952, 2022-12-11:
@RickzFPV The Octo4A needs to download some packages like Python, drivers and many other things to be able to run the OctoPrint itself. The setup script, which i suspect is this one _https://github.com/feelfreelinux/octo4a/blob/master/scripts/setup-octo4a.sh_  used to worked fine, but probably recent Python package update screwed up some dependency and you need to update it manually. To make it working, you need to run the script, then you run the fix, and to make it complete you run the script again.

micah-simmler, 2022-12-11:
> > Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!
> 
> Can confirm this worked for me as well.

Worked for me! Thanks

zdog90210, 2022-12-13:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

This was the fix for me! the user to login is root

RickzFPV, 2022-12-13:
Yeah it worked for me also perfectly on my Lenovo Tablets but I can’t get it to work on my Blu View 2 phones.  Don’t know if the problem is they are not rooted or what but I gave up on them.  Gonna see if I can find some older phones with android 10 on them and see if I can get a few more setup.

KevinDW007, 2022-12-15:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

Can confirm, worked for me on Samsung Galaxy S6 (not rooted)

RickzFPV, 2022-12-15:
Yeah it worked fine for me to on both of my Lenovo tablets running android 10, but haven’t had any luck with the Blu View 2’s running Android 11

RickzFPV, 2022-12-17:
Just Curious has anyone been able to do this fix with a device running Android 11 if so which device.

micah-simmler, 2022-12-17:
I had it running on a galaxy A12 on Android 11 before I updated it.

RickzFPV, 2022-12-17:
Yeah that is when it broke on mine too I didn’t know any better and updated them is there an easy way to reset an android device to before it updated

BCorbett0312, 2022-12-18:
> > Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!
> 
> This was the fix for me! the user to login is root

How did you login as root?  I am only ever able to login using the username octoprint.

RickzFPV, 2022-12-18:
On my tablets I just logged in straight from the up address with port 8022 and it let me right in.  I think android defaults to port 22, but then you don’t have access to octoprint

sanjiv204, 2022-12-19:
https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

sanjiv204, 2022-12-19:
https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

See this it is the solution for this problem.

RickzFPV, 2022-12-19:
Yeah that solution worked on my tablets with windows 10 but it won’t work on the 2 Blu B131DL phones I have running windows 11.

fresh3dprints, 2022-12-20:
This still won't work with my Blu phone that previously had no problem

Nimrodxv230, 2022-12-23:
Is It normal to receive these errors when following t fixed script

```
        return Version(version)
      File "/usr/lib/python3.9/site-packages/packaging/version.py", line 197, in __init__
        raise InvalidVersion(f"Invalid version: '{version}'")
    packaging.version.InvalidVersion: Invalid version: '/tmp/pip'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/5a/03/459c9a4565cabf02d151d0392dc26ef714e3730ac349b4cd0dc27e338287/tornado-3.0.1.tar.gz#sha256=1d0f13df1358ba8f0511bf2d2f7b2cf74abb46f43d45974f8f0797bfcbfbe8f3 (from https://pypi.org/simple/tornado/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading tornado-3.0.tar.gz (348 kB)
     |████████████████████████████████| 348 kB 6.0 MB/s
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-kfnh9pdl/tornado_809a3acec58d42ba8e00d49e4339c393/setup.py'"'"'; __file__='"'"'/tmp/pip-install-kfnh9pdl/tornado_809a3acec58d42ba8e00d49e4339c393/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-3lm2a58y
         cwd: /tmp/pip-install-kfnh9pdl/tornado_809a3acec58d42ba8e00d49e4339c393/
    Complete output (31 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.9/site-packages/setuptools/__init__.py", line 16, in <module>
        import setuptools.version
      File "/usr/lib/python3.9/site-packages/setuptools/version.py", line 1, in <module>
        import pkg_resources
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3243, in <module>
        def _initialize_master_working_set():
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3226, in _call_aside
        f(*args, **kwargs)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
        working_set = WorkingSet._build_master()
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 559, in _build_master
        ws = cls()
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 552, in __init__
        self.add_entry(entry)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 608, in add_entry
        for dist in find_distributions(entry, True):
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2059, in find_on_path
        path_item_entries = _by_version_descending(filtered)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2029, in _by_version_descending
        return sorted(names, key=_by_version, reverse=True)
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in _by_version
        return [packaging.version.parse(part) for part in parts]
      File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2027, in <listcomp>
        return [packaging.version.parse(part) for part in parts]
      File "/usr/lib/python3.9/site-packages/packaging/version.py", line 52, in parse
        return Version(version)
      File "/usr/lib/python3.9/site-packages/packaging/version.py", line 197, in __init__
        raise InvalidVersion(f"Invalid version: '{version}'")
    packaging.version.InvalidVersion: Invalid version: '/tmp/pip'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/94/28/496d3eb40f3f5a3c315f94830c729f2dff00e637b560ed29d8fd48f374da/tornado-3.0.tar.gz#sha256=f5740cd779caa1effb0e6f079433c97bc1b7948c88b296279b63faa463b4222d (from https://pypi.org/simple/tornado/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading tornado-2.4.1.tar.gz (348 kB)
     |████████████████████████████████| 348 kB 6.7 MB/s
```

dzidza, 2022-12-27:
Same problem on Samsung S8

Sejkorka, 2022-12-28:
> > Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!
> 
> Can confirm this worked for me as well.

Hi, can someone please explain me how to do this? I'm a beginner and I don't know how to get there and what the individual steps mean. It would be great if you could provide me with detailed instructions.
Thank you so much

RickzFPV, 2022-12-28:
You probably don’t want to mess with it right now until they release a new version that works correctly.  You need to know how to ssh into the device and add and update things, and if you are brand new it will be extremely difficult.

Sejkorka, 2022-12-28:
> You probably don’t want to mess with it right now until they release a new version that works correctly. You need to know how to ssh into the device and add and update things, and if you are brand new it will be extremely difficult.

If you have time now, I would like to do it now, I know some basics, but I am lost in the procedure that someone mentioned here.

YendorGit, 2023-01-04:
I have only been able to install the 0.0.5 release, without a corrupted notice. I previously had this working, but a reinstall broke it. But, It won't launch the UI. In settings, there is no entry for Server Port. Is that to be the same as the main page that lists 198.162..... etc? If so, It won't allow entry of the dot between the numerals?

zenmaster24, 2023-01-06:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

Confirmed working for me

Jogiarea, 2023-01-12:
325 US-Dollar (Kommentar)
Works for me on Huawai P8 and Ulefon Armor X6 THX

SaulSNES, 2023-01-20:
I understand the steps laid out in the fix that's working for everyone, but I can't for the life of me login on putty or bitvise to be able to start running the commands. Anyone else having an impossible time with SSH???

9cento, 2023-01-20:
> @RickzFPV The Octo4A needs to download some packages like Python, drivers and many other things to be able to run the OctoPrint itself. The setup script, which i suspect is this one _https://github.com/feelfreelinux/octo4a/blob/master/scripts/setup-octo4a.sh_ used to worked fine, but probably recent Python package update screwed up some dependency and you need to update it manually. To make it working, you need to run the script, then you run the fix, and to make it complete you run the script again. Running the script first time is done by the Installer, but as it crashes you have to apply the fix and run it manually again.

Sorry but which fix? And how do I run it without ssh or ttyd (both broken)? I have root access and Termux if it could help somehow

wtaelewis, 2023-01-23:
@RickzFPV thanks for keeping this thread alive. This is a great app, if we could just get someone to fix the current issues. As others have said, you can get it working on Android 10 and lower using the fix found here: https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

Android 11 will not allow SSH to start so you can't apply the fixes above. That's why devices like the BLU View2 B130DL (Android 10) work with the fix applied, but the BLU View2 B131DL (Android 11) do not allow you to process the steps.

2JayR11, 2023-01-24:
> Hey guys, try the fix from here [#325 (comment)](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) . Fixed it for me!

The fix in #325 worked for me but I had to skip step 6 otherwise it wouldn't complete and I had to do it twice and before the second time I rebooted my device.

jschleiden, 2023-01-24:
> unzip octoprint.zip results in no such file

@Chefschweisser were you able to find a fix/workaround? I'm running into the same issue. There doesn't appear to be any file downloaded to my Nexus running Octo4a named "octoprint.zip".

`localhost:~# curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s`
`fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/aarch64/APKINDEX.tar.gz`
`fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/aarch64/APKINDEX.tar.gz`
`OK: 144 MiB in 134 packages`
`Collecting packaging`
`  Using cached packaging-23.0-py3-none-any.whl (42 kB)`
`Installing collected packages: packaging`
`Successfully installed packaging-23.0`
`Collecting https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip`
`  Using cached https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip (6.3 kB)`
`  Installing build dependencies ... done`
`  Getting requirements to build wheel ... done`
`    Preparing wheel metadata ... done`
`localhost:~# unzip octoprint.zip`
`unzip:  cannot find or open octoprint.zip, octoprint.zip.zip or octoprint.zip.ZIP.`

jstarnet, 2023-01-24:
> unzip octoprint.zip results in no such file

@Chefschweisser were you able to find a fix/workaround? I'm running into the same issue. There doesn't appear to be any file downloaded to my Nexus running Octo4a named "octoprint.zip".

`localhost:~# curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s`
`fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/aarch64/APKINDEX.tar.gz`
`fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/aarch64/APKINDEX.tar.gz`
`OK: 144 MiB in 134 packages`
`Collecting packaging`
`  Using cached packaging-23.0-py3-none-any.whl (42 kB)`
`Installing collected packages: packaging`
`Successfully installed packaging-23.0`
`Collecting https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip`
`  Using cached https://github.com/feelfreelinux/octo4a-argon2-mock/archive/main.zip (6.3 kB)`
`  Installing build dependencies ... done`
`  Getting requirements to build wheel ... done`
`    Preparing wheel metadata ... done`
`localhost:~# unzip octoprint.zip`
`unzip:  cannot find or open octoprint.zip, octoprint.zip.zip or octoprint.zip.ZIP.`

2JayR11, 2023-01-24:
@jstarnet 
Have you tried restarting your device and skipping step 6? 
I was getting that same message until I skipped step 6.

9cento, 2023-01-24:
@feelfreelinux is there a way to access octo4a's console without ssh or ttyd? Like Termux, but how should one go about telling Termux where to send those commands? Even only fixing ssh should suffice, the rest could be done through it.

BelowAvgTico, 2023-01-24:
Hello All, 

A complete Noob here, trying to make Octo4a to work. I got the installation corrupted error every time I install the app on the phone I bought for this specifict purpose (Samsung A23). 

I was following the steps found [here ](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) and at first while i am runing the: _pip3 install --upgrade setuptools wheel tornado --no-cache-dir_ command I was getting an error while "Preparing wheel metadata" 

Then it looked like it started to download older tornado versions one after the other until it gives these final 2 errors: 

- Could not find a version that satisfies the requierement tornado
- No matching distribution found for tornado

After doing some updates to python and following some other steps I am now able to run that command with no issues but now I am getting an error following the next step: _curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s_

I am including an screenshot of the error that I am getting now. 
[image]

I feel that I am way out of my technical confort zone trying to get this resolved and I really apreciate any help to get Octo4a working. 

Thanks in advance! 

PS: Sorry if I am not explaining myself clearly or for any grammar error, english is not my first language

rfnovo, 2023-01-24:
Facing the same error on a Galaxy S20 FE. 

> Hello All,
>
> A complete Noob here, trying to make Octo4a to work. I got the installation corrupted error every time I install the app on the phone I bought for this specifict purpose (Samsung A23). [...]

BelowAvgTico, 2023-01-24:
> 

Thank you so much for the help, sadly I do not know how to connect via website, if there is a link on how to do it please let me know. 

Thanks!

BelowAvgTico, 2023-01-24:
> Was facing the same issue and here is how I solved:
> 
>     1. in the app, go do Extensions and enable Remote Web Terminal. Connect using a browser and use root/octoprint as user/password
> 
>     2. now execute the steps mentioned in the article.
> 
> 
> > Hello All,
> > A complete Noob here, trying to make Octo4a to work. I got the installation corrupted error every time I install the app on the phone I bought for this specifict purpose (Samsung A23).
> > I was following the steps found [here ](https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359) and at first while i am runing the: _pip3 install --upgrade setuptools wheel tornado --no-cache-dir_ command I was getting an error while "Preparing wheel metadata"
> > Then it looked like it started to download older tornado versions one after the other until it gives these final 2 errors:
> > 
> > * Could not find a version that satisfies the requierement tornado
> > * No matching distribution found for tornado
> > 
> > After doing some updates to python and following some other steps I am now able to run that command with no issues but now I am getting an error following the next step: _curl -s https://raw.githubusercontent.com/feelfreelinux/octo4a/master/scripts/setup-octo4a.sh | bash -s_
> > I am including an screenshot of the error that I am getting now. [image]
> > I feel that I am way out of my technical confort zone trying to get this resolved and I really apreciate any help to get Octo4a working.
> > Thanks in advance!
> > PS: Sorry if I am not explaining myself clearly or for any grammar error, english is not my first language

Thanks for the help! Sadly I do not know how to connect using a web browser, if you have a link with a tutorial on how to do ot I highly apreciated. 

Thanks!

jstarnet, 2023-01-24:
> @jstarnet Have you tried restarting your device and skipping step 6? I was getting that same message until I skipped step 6.

@2JayR11 I did try restarting. Since I had already run step 6 previously, I tried to pick up on step 7 but I still get the same message:
`localhost:~# unzip octoprint.zip`
`unzip:  cannot find or open octoprint.zip, octoprint.zip.zip or octoprint.zip.ZIP.

I don't see that file (octoprint.zip) in the "fix" and it doesn't make sense to me that you would be required to run the unzip command for a file that isn't there. Feel like something is missing in the fix steps or in where I am trying to execute the commands.
If I skip ahead to step 8 I get the same message as before:
`localhost:~# cd Octo* && pip3 install .`
`-ash: cd: can't cd to Octo*: No such file or directory`

I'm in /root and there is no dir that is created under "Octo". There is the hidden /root/.octoprint/ dir but that doesn't do anything either...

9cento, 2023-01-24:
> I am including an screenshot of the error that I am getting now. [image:
> Octo4a error]

Remote web terminal doesn't work either on Android 11, what version of
Android are you running?

BelowAvgTico, 2023-01-24:
I am using a Samsung A23 with One UI version 4.1 and Android version 12

rfnovo, 2023-01-25:
12

rfnovo, 2023-01-25:
> […](#)

12

chadrebuck, 2023-01-26:
My moto g7 (i think that is the one I have) doesn't have pip3 so I can't do the manual fix I guess.  Any suggestions?  Thanks.

RickzFPV, 2023-01-26:
@chadrebuck when you are doing step 5 are you doing the commands one at a time or all at once I was getting that error with my Samsung phones trying to run them 1 at a time when I ran them all at one time it worked flawless

chadrebuck, 2023-01-26:
I was going to follow the steps @ https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

but found that pip3 wasn't installed when attempting to run

pip3 install --upgrade setuptools wheel tornado --no-cache-dir

RickzFPV, 2023-01-26:
Yeah my Samsung phones didn’t have it either but when I ran step 5 it installed it

RickzFPV, 2023-01-26:
What I did was as soon as I connected through ssh I ran update and upgrade on the packages already on the phone then ran step 5 and it worked perfectly

chadrebuck, 2023-01-26:
thanks - what command are you using to do the update of packages?  I'm familiar with linux but not as much on android.  Does it matter that I'm using a different app to ssh into my phone?  I'm using sshelper because the ssh server from octo print did not work.  Wonder if I get different privileges or app access if the octo print ssh server was working.

RickzFPV, 2023-01-26:
@chadrebuck I use putty to ssh into devices.  As far as the update and upgrade commands mine are on alpine so command is APK update APK upgrade but yours may be different depending on which Linux you have on your phone

chadrebuck, 2023-01-26:
I am using a stock moto g7  running android 10

u0_a216@localhost:SDCard$ uname -a
Linux localhost 4.9.206-perf+ #1 SMP PREEMPT Wed Jan 27 11:44:53 CST 2021 aarch64 Android

[11 more replies]