feelfreelinux/octo4a #329: Need a little help with BLU View 2

RickzFPV, 2022-12-12:
So after doing some extensive research I installed termux on this BLU View 2 phone and I can finally ssh into it but now when I go to install pip3 it won't let me is there something I need to do different with Termux that I didn't have to do with my lenovo Tablet.  My tablet is running perfectly with octo4A and would like to get this phone running the same.  Any help will be much appreciated.

RickzFPV, 2022-12-12:
I am wondering if someone can help me I am able now to ssh into the blue view 2 phone everything communicates fine but when I enter in the commands for the fix it can't find the packages

thomasbra1, 2022-12-12:
If you have the B131DL (Android 11) it won't install and just give a corrupted message.   I have a B130DL working fine with A10 but the B131DL with A11 won't install properly.  Been keeping a eye here to see if a newer version will pop up with a fix.

RickzFPV, 2022-12-13:
Yeah Mine are the B131 DL's and mine are doing the same thing and not installing properly I am gonna just have to wait on these also in hope of a fix or a new version release to use them.  for now I am just using them as touch interfaces on the 2 printers I have raspberry PI's installed on.

cjsz450, 2023-02-28:
Any word on an update for the B131DL?

RickzFPV, 2023-02-28:
I have no idea I was never able to even ssh into one on android 11

alufers, 2023-03-28:
Should work with 1.2.4

closed 2023-03-28 by alufers