feelfreelinux/octo4a #336: This APP is a goner

Xcion54, 2022-12-27:
Tried on 5 different phone models not a single one works, will not install anything will not complete downloads, every download of octoprint is corrupted. from a great app to a useless apk that doesn't do anything other than crash 2 of my phones. Sad, at least bug reporting works 😂

gietl, 2022-12-27:
Have you tried these steps as a work around? https://github.com/feelfreelinux/octo4a/issues/325#issuecomment-1345426359

sashokbg, 2022-12-28:
Hello @Xcion54 this is a community driven project and everyone tries to contribute with whatever they can.

Please refer from sarcastic comments and unproductive issues like "This APP is a goner" as this only demotivates people who already try hard to spend an hour or two after work so that people can enjoy a hobby.

If you can provide some logs, details on phone models and additional information people might be able to help you.

Also please check the comment from @gietl - people have recently worked on some possible solutions to the problems you might have.

RickzFPV, 2022-12-28:
@sashokbg I am curious myself because I have 2 of the BLU View 2 Model # B131DL phones running Android 11 and the fix will not work on them no matter what I try I had it running with the fix, on 2 of my lenovo tablets running Android 10 (but have since moved those tablets to my Klipper Printers to run klipper Screen), With all that has been going on with this lately no one has ever come out and clearly stated if this issue is with Octo4A or Android 11 or a combination of both.  For me personally it is a setback, because I invested a ton of money in equipment to use this setup on printers for my print farm.  I figured myself if this is supported enough I will sit back and wait for a new release that fixes these issues or I will just chalk it up as a loss and lesson learnd.  For most people Octoprint is their only option as they are running Marlin firmware on their printers, with Klipper there are multiple options.

sashokbg, 2022-12-29:
Hello @RickzFPV let's keep the discussion in the other issue and try to involve the project's maintainers

alufers, 2023-03-21:
Sorry to hear that. Sadly neither I or @feelfreelinux has nearly enough time to maintain this app.

closed 2023-03-21 by alufers